arduino-libraries/ArduinoBLE #1: Add a UUID generator function

tigoe, 2018-11-14:
Every BLE sketch needs a handful of them. We all end up going to https://www.uuidgenerator.net/ to get them. Why not just build in the function?

const char[] createUUID(), for example.

sandeepmistry, 2018-11-14:
Hmm, I'm not sure if this is a good idea, https://www.uuidgenerator.net/ mentions:

> What is a Version 1 UUID?
>
> A Version 1 UUID is a universally unique identifier that is generated using a timestamp and the MAC address of the computer on which it was generated.

We won't have easy access to the timestamp unless we connect via WiFi and use NTP.

Then we'd also need to know the devices MAC (or BT address).

> What is a version 4 UUID?
>
> A Version 4 UUID is a universally unique identifier that is generated using random numbers. The Version 4 UUIDs produced by this site were generated using a secure random number generator

Unless we use the crypto chip (which needs to be configured and locked), the Arduino random number API's are pseudo random.

What do you think?

tigoe, 2018-11-14:
Hmmm...

I like the idea of making it easy to generate UUIDs. At the very least, mentioning the uuidgenerator site in the docs is good, perhaps along with these two descriptions. 

It's not possible to use the WiFi and the BLE at the same time, is it?

I suppose it'd be possible to use the RTC and the MAC address to generate the version 1 UUID. But that'd be better as its own example or library.

Other than that, maybe you are right, better to use an easily accessible external source.

t.

sandeepmistry, 2018-11-15:
> I like the idea of making it easy to generate UUIDs. At the very least, mentioning the uuidgenerator site in the docs is good, perhaps along with these two descriptions.

Agreed, when the docs are updated, I'll make sure it gets added

> It's not possible to use the WiFi and the BLE at the same time, is it?

Currently no, the SPI pins connected to the NINA are converted to another UART to use for HCI communications.

tigoe, 2018-11-16:
Hm. Then best to go with a link in the documentation.

tobozo, 2018-11-16:
would [this](https://github.com/1337ninja/UUIDGenerator) fit the need of generating UUIDs ?

mirrorshot, 2021-01-31:
I was wondering: how do you see a compile-time directive to generate a random UUID?
This could have access to system time.