arduino-libraries/ArduinoBLE #3: WiFi Rev2 RGB control

sabas1080, 2018-12-09:
Please add API to control the built-in RGB which is connected to pins 25,26,27 for Arduino MKR and Arduino WiFi Rev2

Thanks

facchinm, 2018-12-10:
Hi @sabas1080 , do you need to control the RGB LED from BLE? If using WiFI, please refer to  https://github.com/arduino-libraries/WiFiNINA/issues/24 for instructions. 
BLE is a bit different since we need a side channel to communicate with ESP32 internally (all the "normal" communication is redirected to BLE "core"). @sandeepmistry any idea on how we could solve it?

sabas1080, 2018-12-10:
Thanks @facchinm

sandeepmistry, 2018-12-10:
Hi @sabas1080,

I think we are going to hold off on implementing something for a while.

Currently, when the NINA is in Bluetooth mode, the ESP32 firmware puts the module in HCI passthrough mode using the standard UART over HCI protocol.

On the MKR WiFi 1010 the ACK and GPIO0 pins and be used for communications but we can't have SPI communications, and on the Uno WiFi rev2 wth CS pin is free. We could consider to have a serial protocol (TX only) on one of the pins to send RGB values. However, then this should be left out of this library and it would create some divergence for the way things would work when in WiFi mode.

sabas1080, 2018-12-10:
Hi @sandeepmistry 

ok, i am trying with WifiNINA but not working led RGB, my sketch is

https://github.com/sabas1080/KanoWandHack/blob/master/arduino-wifi-rev2/arduino-wifi-rev2.ino#L132

sandeepmistry, 2018-12-10:
I suggest trying to add a WiFi.status() call before:

```
  WiFi.status(); <-- added

  WiFiDrv::pinMode(25, OUTPUT);  //GREEN
  WiFiDrv::pinMode(26, OUTPUT);  //RED
  WiFiDrv::pinMode(27, OUTPUT);  //BLUE
  WiFiDrv::analogWrite(25, 128);  // for configurable brightness
  WiFiDrv::analogWrite(26, 128);  // for configurable brightness
  WiFiDrv::analogWrite(27, 128);  // for configurable brightness
```

Once you enable BLE, there is no way to control the LEDs and they will be turn off until WiFi mode is enabled, as ArduinoBLE and WiFiNINA libraries are not compatible.

sabas1080, 2018-12-10:
then I can not use BLE and WIFI at the same time?

sandeepmistry, 2018-12-10:
> then I can not use BLE and WIFI at the same time?

Correct, as stated in the forum: http://forum.arduino.cc/index.php?topic=579469.0

>  Please note, WiFi functionality (using the WiFiNINA library) cannot be used at the same time as BLE.

sabas1080, 2018-12-10:
ok, I hope later you can control, thanks