arduino-libraries/ArduinoBLE #11: ScanResponse Data

dwilday, 2019-08-06:
Can you add support for sending Scan Response Data?

NimaaZx, 2019-08-29:
Hey. Why don't use Ble 5 feature, Secondary advertising channel method instead of scan response and request method  ?

Smigg92, 2019-09-03:
With the scan response is their a method to get the advertised data?

Once I can get that adding additional data, or decoding depending on type of advertisement will be possible?

sandeepmistry, 2019-09-03:
Hi @dwilday,

Is there something specific you want to put in the scan response, in the sense should we have a nice friendly API to do that instead of a generic API for raw scan response data?

For a work around for now you could try to use `HCI. leSetScanResponseData(length, data);`: https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/HCI.h#L53

marcrupprath, 2020-07-15:
Hello,
i tried, but it seams that is will be ignored ?
Dou you have a working example for cutom SCAN response data

Polldo, 2020-07-24:
Hi @marcrupprath , 
a working example for scan response data is already in this library: https://github.com/arduino-libraries/ArduinoBLE/blob/8a49b8c1e0ac195db38a0639378a1b50b7bdd566/src/utility/GAP.cpp#L148
As you can see, currently the local name of the device is not directly advertised but it is sent as a scan response data.

marcrupprath, 2020-07-25:
To make a long story short i will tr direct you to the discussion:
What are the requirements to use , in our case. the Arduino BLE33 as compatible device for LEGO apps.
We did it for ESP32 as you can see: Think about what is missing to do same with ArduinoBLE

https://www.eurobricks.com/forum/index.php?/forums/topic/179180-powered-up-hub-emulator-with-esp32/
https://github.com/GianCann/ATS/blob/master/ESP32_LEGO/PUPHubEmulator.ino

Have a special Look for advertising data, local name and manucaturer data used in this example.

Polldo, 2020-07-27:
Hi @marcrupprath ,
currently it is only possible to insert the local name into the scan response data. I will work on extending the possible parameters in the scan response.
However, as a temporarily workaround, you can directly use the `HCI.leSetScanResponseData` function in this way:
```C++
#include "ArduinoBLE.h"

#define SERVICE_UUID "00001623-1212-EFDE-1623-785FEABCD123"

uint8_t scanData[] = {0x02,0x01,0x06,0x09,0xFF,0x97,0x03,0x00,0x41,0x07,0xB2,0x43,0x00};        

void setup()
{ 
  BLE.begin();
  
  // exploit the already existing functions to advertise the service
  BLE.setAdvertisedServiceUuid(SERVICE_UUID);
  BLE.advertise();
  
  // disable advertising to insert the custom scan response data
  HCI.leSetAdvertiseEnable(0x00);
  HCI.leSetScanResponseData(sizeof(scanData), scanData);
  
  // enable advertising
  HCI.leSetAdvertiseEnable(0x01);
}

void loop()
{

}

```
Also, to access HCI functions you should add `#include "utility/HCI.h"` in 'ArduinoBLE.h' file

Polldo, 2020-11-02:
@marcrupprath @dwilday 
It is now possible to configure both advertising and scan response data given an already encoded raw data packet.
Also scan response data can be configured in a more flexible way
Implemented here https://github.com/arduino-libraries/ArduinoBLE/pull/117
Examples here https://github.com/arduino-libraries/ArduinoBLE/pull/134

closed 2020-11-02 by Polldo