arduino-libraries/ArduinoBLE #19: BLE Serial Example

hongkai0818, 2019-09-01:
Hi Possible to add an example on ble serial similar to
https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/master/examples/serial

Thanks

sandeepmistry, 2019-09-03:
@tigoe any opinions on this?

tigoe, 2019-09-03:
I don't love the idea, since BLE isn't serial, and it's confusing to users who think it should show up on their laptop/desktop machines as a serial port. Unless you have an app to pair it with that'll provide a "serial" connection, I think it'll only be confusing.

OTOH, if BLE 4 and 5 now supports real serial, and it can actually show up as a COM port on MacOS and Windows and Linux without having to modify the central device's OS, I am all for it.

hongkai0818, 2019-09-04:
Hi, thanks for your reply.
The reason I was asking that is because I have previously explored NRF52DK and Bluefruit NRF52. Both SDK provides examples on both BLE central and peripheral that are exposing to BLEUART (NUS).  Current examples in the arduinoBLE do not have the BLEUART example using NUS. That might be good to add BLEUART examples in the future release of the library.

kundk-ckret, 2019-09-19:
Is it even possible to create a serial connection via regular BT 5 on the nano 33 ble sense?

I'm currently searching for a way to Transfer  bulk data and had no luck so far :(

tigoe, 2019-09-19:
How big is "bulk"?  If I remember correctly, BLE transfers data in
something like 20-byte packets, so for something like a file transfer,
you'd need to break it up into those chunks. If you were making a file
transfer API, you'd make it break the file up and packetize it.  OTOH, if
you're talking about transfer of sensor data where each reading or group of
readings is less than 20 bytes total, it's no problem. The BLE spec would
have you make a characteristic for each sensor reading, and that's what's
easiest in the API, but you could presumably make a string or array with a
bunch of numbers in it. That's how BLE MIDI works, for example.

kundk-ckret, 2019-09-20:
Well... I'm building a data logger for model rockets...
The Arduino nano 33 Sense had the perfect fit for me, but i thought that BLE might be the best idea to get the data from the Arduino after recovery...

so we're talking about not more than a few KB if done correctly i think.

I'm currently experimenting with the notification feature to push the values through, but it feels like im abusing it...

tigoe, 2019-09-20:
That's pretty much what notification feature is for, so don't worry about
it.

T

kundk-ckret, 2019-09-20:
one last question (and i know i'm abusing this topic right now :D)

I have a struct for my logging data points - BLECharacteristic are typed... the largest thing i can get is BLELongCharacteristic or something like that, but I want to transfer my struct as a byte array... 

how would I manage that?
I think, there will be others with this problem...

tigoe, 2019-09-21:
Check out the ble midi examples in @sandeepmistry, @don, and
@alasdairallan's book "make: Bluetooth". They use byte arrays. Here's an
example derived from them:
https://github.com/tigoe/SoundExamples/blob/master/MIDI_examples/ArduinoBLEMIDI/ArduinoBLEMIDI.ino

kundk-ckret, 2019-09-22:
Thank you all for your help - got it all working.

For anyone who is interested - my code is in https://github.com/adastra-rocketry/adastra-telemetry/tree/master/ArduinoNano33BLESense/Telemetry