arduino-libraries/ArduinoBLE #22: Mesh

ixj111, 2019-09-04:
Hi, is there any plan to add the support for BLE mesh? Thanks.

JWGmeligMeyling, 2019-09-04:
It doesn't seem so: https://github.com/arduino-libraries/ArduinoBLE/issues/6#issuecomment-485814549

charly37, 2019-09-11:
I m also very interested in the future support. Wonder if anybody actively working on this library has any kind of insider info to share ? BTW I just manage to have my nano BLE33 working fine with this lib and it was very easy and well documented so big thanks to the DEV team !

charly37, 2019-09-12:
it seems that MESH network in BT are quite complicated.... some doc here https://www.novelbits.io/bluetooth-mesh-tutorial-part-3/ explaining the notion of provisionning with a "special" device "device used to add a node to the network is called the provisioner (usually a tablet, smartphone, or a PC)." and most of the doc example online rely on some SDK :( No idea how much work it would be to have it supported in this lib

antlomb, 2020-11-02:
Any news one year later ?

electroc, 2022-04-20:
Any news 1.5 years later ?

0x44454c, 2024-05-16:
Is there a way to develop BLE mesh network based sensor data collection in Arduino BLE 33? Searched a lot of libraries especially nRF sdk, and mbed mesh network, but couldn't figure out how to implement BLE mesh.

Require urgent help.

ltrush, 2025-07-10:
Any update on this? @charly37 @0x44454c did either of you figure anything out?