arduino-libraries/ArduinoBLE #25: Best way to observe a peripheral’s RSSI

stapelberg, 2019-09-07:
I want to build a key bowl which detects whether a certain keychain was placed in it by detecting the attached Tile tag via BLE.

What is the most energy-efficient yet reliable way to observe a peripheral’s RSSI?

Currently, I am scanning repeatedly, limited to 4x per second and the specific peripheral address I am interested in:
```c
#include <ArduinoBLE.h>

void setup() {
  Serial.begin(9600);
  while (!Serial); // wait
  
  Serial.println("setup");
  BLE.begin();
  BLE.setEventHandler(BLEDiscovered, discovered);
  BLE.scanForAddress("xx:yy:zz:aa:bb:cc", true);
}

void logDevice(BLEDevice dev) {
  Serial.print("BLE device found: ");
  Serial.print("address=");
  Serial.print(dev.address());
  Serial.print(", rssi=");
  Serial.print(dev.rssi());
  if (!dev.hasLocalName()) {
    Serial.println("");
    return;
  }
  Serial.print(", name=");
  Serial.print(dev.localName());
  Serial.println("");
}

void discovered(BLEDevice dev) {
  BLE.stopScan();
  logDevice(dev);
  delay(250); // ms
  BLE.scanForAddress("xx:yy:zz:aa:bb:cc", true);
}

void loop() {
  BLE.poll(1000); // ms
}
```

I’m a bit hesitant to connect to the device, because I don’t want to disrupt any other functionality of the Tile.

Is there a better way than the approach displayed above?

Thanks,

sandeepmistry, 2019-09-09:
Hi @stapelberg,

Your sketch looks fine, are you planning to power your central via a battery?

If you are, and using a Arduino Nano 33 BLE board, please follow pull request #15 for more low power tricks.

stapelberg, 2019-09-10:
Thanks for taking a look! Yes, I am planning to power the nano 33 BLE from a battery. Thanks for the pointer to the power saving PR :)

closed 2019-09-10 by stapelberg