arduino-libraries/ArduinoBLE #27: BLE central read characteristic takes 400 m/s

HansAMpaD, 2019-09-13:
Not sure if this is the right place to ask, tried de Arduino Forum without succes. 

With a Arduino Nano 33 IOT, as central device, I'm able to connect with my Ultrasonic anemometer, discover services (180d) and subscribe to the characteristic (2a39) which holds the value with the wind data. Each time when I read this value it takes 400 milliseconds (where it sends at 4 Hz) between the results. The code is based on the Peripheral Explorer example. 

The characteristic has read and notify capability. Think I'm sending a "read" each time instead of responding on notification. Tried setEventHandler, the value is 1 but the length is 0 (where with read the length was 10). 

The developers instruction from the Ultrasonic:
https://calypsoinstruments.com/website/static/Dev_manual_Ultrasonic_Portable_1.8.pdf

Hope someone can give some guidance. Thanks in advance.

sandeepmistry, 2019-09-18:
Hi @HansAMpaD,

Could you please provide additional details, including:

1) Which version of the library you are using

2) The output of the PeripherExplorer sketch, after you've changed it to target the Ultrasonic anemometer

3) A minimal sketch to reproduce the issue

4) The Serial Monitor output of 3) with `BLE.debug(Serial)` placed in `setup()`. Please also enable "Show timestamp" to track the timing.

HansAMpaD, 2019-09-18:
Hi sandeepmistry, thanks for the reply.

I'm using Arduino 1.8.5 (because of Teensy's) and the latest BLE 1.1.1. library
The code:
`#include <ArduinoBLE.h>

byte dataW[10];
unsigned long TimerStart;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  // begin initialization
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");

    while (1);
  }
  BLE.debug(Serial);
  Serial.println("BLE Central - Ultrasonic");

  // start scanning for ULTRASONIC
  BLE.scan();
}

void loop() {
  // check if a ULTRASONIC has been discovered
  BLEDevice ULTRASONIC = BLE.available();

  if (ULTRASONIC) {
    // discovered a ULTRASONIC, print out address, local name, and advertised service
    Serial.print("Found ");
    Serial.print(ULTRASONIC.address());
    Serial.print(" '");
    Serial.print(ULTRASONIC.localName());
    Serial.print("' ");
    Serial.print(ULTRASONIC.advertisedServiceUuid());
    Serial.println();

    // Check if the ULTRASONIC is a SensorTag, the local name will be:
    // "CC2650 SensorTag"
    if (ULTRASONIC.localName() == "ULTRASONIC") {
      // stop scanning
      BLE.stopScan();

      connectUltrasonic(ULTRASONIC);

      // ULTRASONIC disconnected, start scanning again
      BLE.scan();
    }
  }
}

void connectUltrasonic(BLEDevice ULTRASONIC) {
  // connect to the ULTRASONIC
  Serial.println("Connecting ...");
  if (ULTRASONIC.connect()) {
    Serial.println("Connected");
  } else {
    Serial.println("Failed to connect!");
    return;
  }

  // discover ULTRASONIC attributes
  Serial.println("Discovering service 0x180d ...");
  if (ULTRASONIC.discoverService("180d")) {
    Serial.println("Service discovered");
  } else {
    Serial.println("Attribute discovery failed.");
[... 110 lines cut ...]
HCI ACLDATA RX <- 02002005000100040013
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
Subscribed
HCI ACLDATA TX -> 0200000700030004000A1B00
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200F000B0004000B00000000067A00000000
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
TimerStart = 14186
HCI ACLDATA TX -> 0200000700030004000A1B00
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
HCI ACLDATA RX <- 0200200F000B0004000B00000000067A00000000
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
TimerStart = 14586
HCI ACLDATA TX -> 0200000700030004000A1B00
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
HCI ACLDATA RX <- 0200200F000B0004000B00000000067A00000000
TimerStart = 14984
HCI ACLDATA TX -> 0200000700030004000A1B00
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
HCI ACLDATA RX <- 0200200F000B0004000B00000000067A00000000
HCI ACLDATA RX <- 02002011000D0004001B1B0000000000067A00000000
TimerStart = 15386

HansAMpaD, 2019-09-18:
Sorry wrong button

HansAMpaD, 2019-09-18:
Downloaded the newest Arduino 1.8.10, so now with timestamps:

19:25:38.072 -> BLE Central - Ultrasonic
19:25:38.072 -> HCI COMMAND TX -> 010C20020001
19:25:38.105 -> HCI EVENT RX <- 040E04050C200C
19:25:38.105 -> HCI COMMAND TX -> 010B200701100010000000
19:25:38.105 -> HCI EVENT RX <- 040E04050B2000
19:25:38.105 -> HCI COMMAND TX -> 010C20020101
19:25:38.105 -> HCI EVENT RX <- 040E04050C2000
19:25:38.622 -> HCI EVENT RX <- 043E1702010001D9B0CFE8EE730B02010607FF4C0010020108AA
19:25:39.148 -> HCI EVENT RX <- 043E2002010001526F36D96EE61402010603030D180C09554C545241534F4E494300C7
19:25:39.148 -> HCI EVENT RX <- 043E0C02010401526F36D96EE600C8
19:25:39.148 -> Found e6:6e:d9:36:6f:52 'ULTRASONIC ' 180d
19:25:39.148 -> HCI COMMAND TX -> 010C20020000
19:25:39.148 -> HCI EVENT RX <- 040E04050C2000
19:25:39.148 -> Connecting ...
19:25:39.148 -> HCI COMMAND TX -> 010D2019600030000001526F36D96EE60006000C000000C80004000600
19:25:39.148 -> HCI EVENT RX <- 040F0400050D20
19:25:42.367 -> HCI EVENT RX <- 043E13010000000001526F36D96EE60C000000C80000
19:25:42.367 -> Connected
19:25:42.367 -> Discovering service 0x180d ...
19:25:42.367 -> HCI ACLDATA TX -> 02000007000300040002F200
19:25:42.367 -> HCI EVENT RX <- 0413050100000100
19:25:42.367 -> HCI ACLDATA RX <- 02002010000C000500120208005000A00000009001
19:25:42.367 -> HCI ACLDATA TX -> 0200000A0006000500130202000000
19:25:42.367 -> HCI COMMAND TX -> 0113200E00005000A0000000900104000600
19:25:42.367 -> HCI EVENT RX <- 040F0400051320
19:25:42.400 -> HCI EVENT RX <- 0413050100000100
19:25:42.400 -> HCI ACLDATA RX <- 020020070003000400031700
19:25:42.400 -> HCI ACLDATA TX -> 0200000B0007000400100100FFFF0028
19:25:42.400 -> HCI EVENT RX <- 0413050100000100
19:25:42.400 -> HCI ACLDATA RX <- 020020180014000400110601000700001808000B0001180C0018000A18
19:25:42.433 -> HCI ACLDATA TX -> 0200000B0007000400101900FFFF0028
19:25:42.433 -> HCI EVENT RX <- 0413050100000100
19:25:42.433 -> HCI ACLDATA RX <- 0200200C000800040011061900FFFF0D18
19:25:42.433 -> HCI ACLDATA TX -> 0200000B0007000400081900FFFF0328
19:25:42.471 -> HCI EVENT RX <- 0413050100000100
19:25:42.471 -> HCI ACLDATA RX <- 0200201B001700040009071A00121B00392A1D00021E0001A01F000A200002A0
19:25:42.471 -> HCI ACLDATA TX -> 0200000B0007000400082100FFFF0328
19:25:42.507 -> HCI EVENT RX <- 0413050100000100
19:25:42.609 -> HCI EVENT RX <- 043E0A03000000A00000009001
19:25:42.609 -> HCI ACLDATA RX <- 0200201B0017000400090721000A220003A023000A240007A025000A260008A0
19:25:42.609 -> HCI ACLDATA TX -> 0200000B0007000400082700FFFF0328
19:25:42.818 -> HCI EVENT RX <- 0413050100000100
19:25:43.025 -> HCI ACLDATA RX <- 0200201B0017000400090727000A280009A029000A2A000AA02B000A2C000BA0
19:25:43.025 -> HCI ACLDATA TX -> 0200000B0007000400082D00FFFF0328
19:25:43.230 -> HCI EVENT RX <- 0413050100000100
19:25:43.434 -> HCI ACLDATA RX <- 0200200D000900040009072D00122E000CA0
19:25:43.434 -> HCI ACLDATA TX -> 0200000B0007000400082F00FFFF0328
19:25:43.614 -> HCI EVENT RX <- 0413050100000100
19:25:43.833 -> HCI ACLDATA RX <- 02002009000500040001082F000A
19:25:43.833 -> HCI ACLDATA TX -> 020000090005000400042F00FFFF
19:25:44.012 -> HCI EVENT RX <- 0413050100000100
19:25:44.221 -> HCI ACLDATA RX <- 0200200A000600040005012F000229
19:25:44.221 -> HCI ACLDATA TX -> 020000090005000400043000FFFF
19:25:44.404 -> HCI EVENT RX <- 0413050100000100
19:25:44.612 -> HCI ACLDATA RX <- 020020090005000400010430000A
19:25:44.612 -> Service discovered
19:25:44.612 -> Subscribing to DataInCharacteristic ...
19:25:44.612 -> HCI ACLDATA TX -> 020000090005000400121C000100
19:25:44.816 -> HCI EVENT RX <- 0413050100000100
19:25:45.023 -> HCI ACLDATA RX <- 02002005000100040013
19:25:45.023 -> HCI ACLDATA RX <- 02002011000D0004001B1B0043000000067B00000000
19:25:45.023 -> Subscribed
19:25:45.023 -> HCI ACLDATA TX -> 0200000700030004000A1B00
19:25:45.199 -> HCI EVENT RX <- 0413050100000100
19:25:45.199 -> HCI ACLDATA RX <- 02002011000D0004001B1B0043000000067B00000000
19:25:45.411 -> HCI ACLDATA RX <- 0200200F000B0004000B43000000067B00000000
19:25:45.411 -> TimerStart = 10969
19:25:45.411 -> HCI ACLDATA TX -> 0200000700030004000A1B00
[... 8 lines cut ...]
19:25:46.219 -> HCI ACLDATA RX <- 0200200F000B0004000B3F000000067B00000000
19:25:46.219 -> HCI ACLDATA RX <- 02002011000D0004001B1B0042000000067B00000000
19:25:46.219 -> TimerStart = 11771
19:25:46.219 -> HCI ACLDATA TX -> 0200000700030004000A1B00
19:25:46.431 -> HCI EVENT RX <- 0413050100000100
19:25:46.607 -> HCI ACLDATA RX <- 0200200F000B0004000B42000000067B00000000
19:25:46.607 -> HCI ACLDATA RX <- 02002011000D0004001B1B004A000000067B00000000
19:25:46.607 -> TimerStart = 12170
19:25:46.607 -> HCI ACLDATA TX -> 0200000700030004000A1B00
19:25:46.814 -> HCI EVENT RX <- 0413050100000100
19:25:46.814 -> HCI ACLDATA RX <- 02002011000D0004001B1B0042000000067B00000000
19:25:47.028 -> HCI ACLDATA RX <- 0200200F000B0004000B42000000067B00000000
19:25:47.028 -> HCI ACLDATA RX <- 02002011000D0004001B1B0050000000067B00000000
19:25:47.028 -> TimerStart = 12571
19:25:47.028 -> HCI ACLDATA TX -> 0200000700030004000A1B00
19:25:47.205 -> HCI EVENT RX <- 0413050100000100
19:25:47.205 -> HCI ACLDATA RX <- 02002011000D0004001B1B004E000000067B00000000
19:25:47.409 -> HCI ACLDATA RX <- 0200200F000B0004000B4E000000067B00000000
19:25:47.409 -> TimerStart = 12969
19:25:47.409 -> HCI ACLDATA TX -> 0200000700030004000A1B00
19:25:47.624 -> HCI EVENT RX <- 0413050100000100
19:25:47.624 -> HCI ACLDATA RX <- 02002011000D0004001B1B004E000000067B00000000
19:25:47.830 -> HCI ACLDATA RX <- 0200200F000B0004000B4E000000067B00000000
19:25:47.830 -> HCI ACLDATA RX <- 02002011000D0004001B1B0043000000067B00000000
19:25:47.830 -> TimerStart = 13371

sandeepmistry, 2019-09-18:
```

sandeepmistry, 2019-09-18:
```
19:25:47.028 -> HCI ACLDATA TX -> 020000070003000400 0A1B00 <- read request
19:25:47.205 -> HCI EVENT RX <- 0413050100000100
19:25:47.205 -> HCI ACLDATA RX <- 02002011000D000400 1B1B004E000000067B00000000 <- notification
19:25:47.409 -> HCI ACLDATA RX <- 0200200F000B000400 0B4E000000067B00000000 <- read response
```

So it seems the device takes 400 ms to reply to the read response.

It's not clear, why you are subscribing to the characteristic and then also reading. I would suggestion you either manually read or only listen for subscriptions (see SensorTag example for more details).

HansAMpaD, 2019-09-18:
That was my lack of knowledge, I tried different combinations without succes. 
With:
`if (DataInCharacteristic.valueUpdated()) {
      byte value;
     
      DataInCharacteristic.readValue(value);

   printData(DataInCharacteristic.value(), DataInCharacteristic.valueLength());
   windSpeed = ((int(dataW[1]) * 256) + (int(dataW[0])));
   Serial.print("windSpeed = ");Serial.println(windSpeed);
    }`
Now it's updating as expected. Thanks for pointing the right direction!

closed 2019-09-18 by HansAMpaD