arduino-libraries/ArduinoBLE #28: Setting custom Advertising Data

akikarhu, 2019-09-15:
Creating temperature sensor on Arduino Namo 33 IoT, and to save battery, non-connected broadcast would do fine. Otherwise great library, but I cannot figure out how to set custom Advertising Data (those 31 octets). Maybe using the setLocalName(name)? Am I missing something or has this a feature not yet been implemented? If not, could you please implement something like BLE.setAdvertisingData(byte[])?

sandeepmistry, 2019-09-18:
Hi @akikarhu,

> If not, could you please implement something like BLE.setAdvertisingData(byte[])?

I'm not very keen on exposing this, as people will have to manually EIR encode the array.

However here are some other suggestions:

1) Put the data in the manufacturer data, using `BLE.setManufacturerData(...)` and call stop/start advertising when changing

2) Use service data, you'll need the changes from PR https://github.com/arduino-libraries/ArduinoBLE/pull/29 - I've included a test sketch in the comment of the pull request.

Please let us know your thoughts on this.

Polldo, 2020-11-02:
@akikarhu 
It is now possible to configure both advertising and scan response data given an already encoded raw data packet.
Implemented here #117
Examples here #134

closed 2020-11-02 by Polldo