arduino-libraries/ArduinoBLE #33: BLE nano 33 does not report or disconnect from centra

Russell108, 2019-10-02:
When using for example the button peripheral and central examples in the arduinoBLE library
If for example the central is a MKR1010 powered by usb and the peripheral is a nanoBLE 33 then if I remove the power from the central the peripheral still reports it as connected.

If instead of a Nano BLE 33 I use a Nano IOT33 then when the central is de powered the peripheral reports it as being disconnected.

there are several issues regarding disconnection & this library when using the Nana BLe 33.

Arduino support said the following

================================================================

Hello Russell,

Thank you for reporting us this issue,

We request you to open an issue on Github here and our developers will provide you further guidance.

Have a nice day!

Best Regards,
Sravya Amirisetti

sandeepmistry, 2019-10-03:
@Russell108 thanks for the issue report! I was able to reproduce here.

@facchinm it seems the Cordio stack is not generating the HCI disconnect event when power is cut to the central.

If I add `BLE.debug(Serial);` and using the MKR WiFi 1010 as the BLE peripheral + Nano 33 BLE as the central, on the peripheral side I get the following event:

```
HCI EVENT RX <- 04050400000008
```

... but if the roles are reversed, nothing ...

sandeepmistry, 2019-10-03:
I tried lowering the supervision timeout from 0xc8 to 0x14 on the central side, this did not have any impact.

ixj111, 2019-10-24:
Hello, ant update on this issue? Still exists

harpop, 2019-11-13:
Hello, just received my nano 33 ble devices. Stumbled on the same issue. As always lost a few hours! Any update?

Russell108, 2019-11-13:
Ive gone back to a nano not 33 & wifi

sandeepmistry, 2019-11-29:
Hi @ixj111 @Russell108 @harpop @alexisicte,

Pull request #44 should fix this, if you have some time to try out the changes and provide your feedback that would be great.

closed 2019-12-02 by facchinm

alexisicte, 2019-12-02:
Hi @sandeepmistry,
Before your last changes when an "unnormal" disconnect issue occurs ble debugging returns nothing. After your changes we get on the serial monitor:

```
11:04:18.535 -> HCI EVENT RX <- 04050400000008
11:04:18.535 -> HCI COMMAND TX -> 010A200101
11:04:18.535 -> HCI EVENT RX <- 040E04010A2000
```
When "normal" disconnect procedure occurs the output is:

```
10:55:22.611 -> HCI EVENT RX <- 04050400000013
10:55:22.611 -> HCI COMMAND TX -> 010A200101
10:55:22.646 -> HCI EVENT RX <- 040E04010A2000
```
While `BLE.adverise()` is called the output is the following in a loop:

```
10:54:18.858 -> HCI EVENT RX <- 040E040106200C
10:54:18.858 -> HCI COMMAND TX -> 0106200FA000A0000000000000000000000700
10:54:18.858 -> HCI EVENT RX <- 040E040106200C
10:54:18.858 -> HCI COMMAND TX -> 0106200FA000A0000000000000000000000700
10:54:18.891 -> HCI EVENT RX <- 040E040106200C
10:54:18.891 -> HCI COMMAND TX -> 0106200FA000A0000000000000000000000700
10:54:18.891 -> HCI EVENT RX <- 040E040106200C
```

but, after an "unnormal" disconnect issue occurs the above output is not displayed. Actually the serial monitor is idle. When i try to reconnect after an "unnormal" disconnect issue, i get:

```
11:05:29.198 -> HCI EVENT RX <- 043E13010000000101E397B78CE96128000000F40101
11:05:29.467 -> HCI ACLDATA RX <- 0200200B0007000400100100FFFF0028
11:05:29.834 -> HCI EVENT RX <- 043E0A0300000006000000F401
11:05:34.444 -> HCI ACLDATA RX <- 0200200B0007000400100100FFFF0028
11:05:39.459 -> HCI ACLDATA RX <- 0200200B0007000400100100FFFF0028
```
AND

```
11:05:44.468 -> HCI EVENT RX <- 04050400000013
11:05:44.468 -> HCI COMMAND TX -> 010A200101
11:05:44.468 -> HCI EVENT RX <- 040E04010A2000
```
which is the same as "normal" disconnect output. Then the serial monitor is idle as before.
According to BLE.debug it seems that the device is not advertising because i dont get the advertise debug output, but central devices can see the peripheral.

JGronholz, 2020-05-20:
This issue persists as of 05/20/2020, library version 1.1.2.
If I disconnect power from my Nano 33 Sense peripheral, my Nano 33 Sense central remains unaware and connected.

This event never fires on my Nano 33 Sense:
`BLE.setEventHandler(BLEDeviceEvent::BLEDisconnected, disconnectedHandler);`

From this thread, almost a month ago:
(https://forum.arduino.cc/index.php?topic=645907.0)

Klaus_K appears to have done everything but fix the code:

> The behavior for them is different. I guess they both use the same BLE library but there might be differences in the integration with mbedOS on the Nano 33 BLE.
> 
> The Arduino Nano 33 BLE does not set the central.connected correctly when the connection is lost. It returns central.rssi() = 0 a couple of times when the connection is about to get lost. But it can recover from that when the connection gets better. But when the connection is lost it cannot recover without reset.
> 
> The Arduino Nano 33 IoT does set central.connected correctly so the sketch can get back to a state where it waits for connections. The central.rssi never gets to 0, but gets to some maximum negative value around -128 or so as expected.

Thanks, Jon!

sprasadkrish, 2020-06-14:
Hi 
Hardware - Arduino MKR WIFI1010 (BLE) . using BLE communication .
         We tried interface one central and one peripherals, its working.
but we need to connect multiple peripheral and one central.
Example:
1) central sends request to peripheral-1, peripheral-1 receive requests and send data 
    (example value 1234) to central 
2) central sends request to peripheral-2, peripheral-2 receive requests and send data 
    (example value 5678) to central  

I tried to connect one central and two peripheral, one by one in the same program, 
but it accepts any one peripheral , example peripheral-1 or peripheral-2 .

could you please suggest how to communicate with multiple peripherals?
[image]

bretjackson, 2020-06-19:
I have the same issue with the nano 33 BLE not recognizing when the peripheral disconnects from a lost of power. Any updates?

sprasadkrish, 2020-06-25:
I solved my problem , i used CONNECT METHOD  for connect multiple devices .

sprasadkrish, 2020-06-25:
> Hi
> Hardware - Arduino MKR WIFI1010 (BLE) . using BLE communication .
> We tried interface one central and one peripherals, its working.
> but we need to connect multiple peripheral and one central. [...]

I solved my problem , i used CONNECT METHOD  for connect multiple devices .

sprasadkrish, 2020-06-25:
> 
> 
> I have the same issue with the nano 33 BLE not recognizing when the peripheral disconnects from a lost of power. Any updates?

use CONNECT METHOD instead of Broadcast method .

Polldo, 2020-07-02:
Hi @JGronholz ,
I have tried to reproduce the error you mentioned, but after a power cut-off both the central and the peripheral were correctly triggering the disconnection event.
 
Here's the setup I used:
Arduino Core: Arduino-nRF582x version 1.1.5
ArduinoBLE: version 1.1.2-8-g3b228dc
BLE Peripheral: nano 33 ble
BLE Central: nano 33 ble

Would you mind trying again your sketch using the core and library version I mentioned? 
I leave you as a reference the sketches I used

Central code:
```C++
#include <ArduinoBLE.h>

void blePeripheralConnectHandler(BLEDevice dev) {
    Serial.println("Connected event, peripheral: ");
    Serial.println(dev.address());
}

void blePeripheralDisconnectHandler(BLEDevice dev) {
    Serial.println("Disconnected event, peripheral: ");
    Serial.println(dev.address());
    BLE.scanForName("Test");
}

void setup() {
  Serial.begin(115200);
  delay(500);

  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }

  Serial.println("Central on");

	BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
	BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);

  BLE.scanForName("Test");
}

void loop() {

  delay(1000);
  BLEDevice peripheral = BLE.available();

  if (peripheral) {
    BLE.stopScan();
    Serial.println("found peripheral");

    if (peripheral.connect()) {
    } else { 
      BLE.scan();
    }
  }
}
```

Peripheral code:
```C++
#include <ArduinoBLE.h>

void setup() {
    Serial.begin(115200);
    delay(500);
    if (!BLE.begin()) {
        Serial.println("starting BLE failed!");
        while (1);
    }

    BLE.debug(Serial);

    BLE.setLocalName("Test");
    BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
    BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);
    BLE.advertise();
    Serial.println("Bluetooth device active, waiting for connections...");
}

void blePeripheralConnectHandler(BLEDevice central) {
    Serial.println("Connected event, central: ");
    Serial.println(central.address());
}

void blePeripheralDisconnectHandler(BLEDevice central) {
    Serial.println("Disconnected event, central: ");
    Serial.println(central.address());
}

void loop() {
    static long prvTime = 0;
    BLE.poll();
    if (millis() - prvTime >= 1000) {
        prvTime = millis();
        if (BLE.connected()) {
            Serial.println("---");
            int rssiValue = BLE.rssi();
            Serial.println(rssiValue);
            Serial.println("Connected to central");
        }
        else {
            Serial.println("Not connected to central");
        }
    }
}
```

sprasadkrish, 2020-08-21:
MKR1010 BLE communication :
I am using one central and two peripherals for data receiving.
1) Central always receives data, based on a request to peripherals.
Example: central request peripheral-1, to send data, peripheral-1 send data to central similarly peripheral-2 also.

Issue:
Central receive up to 2 minutes , after two minutes , central disconnect peripheral. there is any possible to increase the timeout of BLE, or possible to set NO TIMEOUT (continues data transfer).

Polldo, 2020-08-21:
Hi @sprasadkrish ,
can you please open a new issue for this?
Also can you please provide information about your setup (library version and boards used) and a test sketch? 
Thanks!

PR #44 (2019-12-02): Cordio: BLE thread loop fixes
This resolves #33.

This pull request has two changes:

1) It calls `timer.start()` in the BLE thread loop. Without this, `timer.read_high_resolution_us()` always returns 0.

**Note:** the issue is also present in the mbed sample code the HCI transport was based on: https://github.com/ARMmbed/mbed-os-cordio-hci-passthrough/blob/fix-wsf-msg/main.cpp#L86 (@facchinm should we inform them?)

2) It reverts part of pull request #15. Unfortunately, the `WSF_MS_PER_TICK` needs to match the value mbed was compiled with for the supervision timeout to match: https://github.c [... cut]