arduino-libraries/ArduinoBLE #37: No low power mode possible

Knowless, 2019-10-11:
Hello everybody,
I read that the BLE part of the processor runs on its own. Now I wanted to put the main processor into sleep mode.
Normally the processor should find sleep/deepsleep by itself if there is nothing to do. However, I still measure currents in the examples of about 16-18mA (Arduino Nano 33 BLE).
So I assume that the sleep-mode is blocked by a function.
Can someone confirm this or give me a clue how to put the controller into sleep mode? I tried the mbed::sleep() command, but without luck too.
It should be possible to send the main core to sleep and maintain the BLE functionality, right?

I miss the register list where I could try to make the changes by my self. Could someone help me and say where the software guide is?

Greetings,
Knowless

facchinm, 2019-10-14:
Hi @Knowless ,
this patch (and comment) explains the current situation:
https://github.com/arduino-libraries/ArduinoBLE/pull/15#issuecomment-530084737

We didn't release any update yet, so to get the patch you'll need the git version of the library.

The biggest problems were:
* the wakelock being held by Cordio
* the inefficient  DC/DC converter

The patch issues the first problem; for the second one we decided to provide a solder jumper on the bottom of the board to allow powering it via 3.3V directly.
This makes reprogramming a bit harder (since the board must always be powered via external 3.3V) but cuts power consumption down to the expected sub-milliampere.

Further reduction can be obtained by removing I2C sensors power and pullups using
```
  digitalWrite(PIN_ENABLE_SENSORS_3V3, LOW);
  digitalWrite(PIN_ENABLE_I2C_PULLUP, LOW);
 ```

Knowless, 2019-10-15:
Hi fcchinm,
Thanks for the feedback and the summary of #15.
I found the thread yesterday but it's pretty neatly summarized.
I will try to implement it soon.

Unfortunately I'm not the expert on how to remove ArduinoBLE, so I can install it from github.
Originally I installed the library via the IDE. 

Therefore an untested guide for others (just to see how it should work(read in other forum)):
Just delete the ArduinoBLE folder under : 
...user\documents\arduino\libraries
Then copy the git version to the same location and the examples were found by the IDE.

If it is okay for you facchinm, I would post this answer in the Arduino forum, because I had already asked the question there weeks ago and I think a lot people are interested in this topic.

When can we expect an official release?

And agein. Thank you.

facchinm, 2019-10-16:
Official release is probably due next week; in the meantime, feel free to cross post on the forum, linking to this issue so if we update something here people will always find the latest information.

ixj111, 2019-10-25:
@facchinm  Hi
I use the online web-based arduino IDE. Do I still need to use git version?

facchinm, 2019-10-28:
Yes, we are fixing some more stuff before releasing 1.1.2, but you can download the library as zip and import in Create to use the version containing the fix.

sandeepmistry, 2019-11-12:
Hi @Knowless @ixj111,

I've just pushed a new 1.1.2 tag which includes the changes. It should be available in both the Arduino IDE library manager and the Create Web IDE in the next few hours.

closed 2019-11-12 by sandeepmistry

ixj111, 2019-11-12:
Great! Thanks

smartboxchannel, 2020-02-24:
Today I downloaded the library. I tried, still without sleep, consumption ~ 7mA