arduino-libraries/ArduinoBLE #38: ibeacon and eddystone

ixj111, 2019-10-24:
Hi, i am wondering if the library can be used to read beacon (eg., ibeacon and eddystone) data?

sandeepmistry, 2019-11-11:
@ixj111 not at this time, there is no way to access manufacturer data or service data from the scanned advertisements.

However, we'd welcome a pull request for these features.

bluelu, 2020-05-10:
Hi, I would also love to see this. Thanks

nico7885, 2020-05-22:
Yes, i would like, this would also make it possible to read the broadcast characteristics of the library.

uasiddiqi, 2020-10-01:
Is there still any way to read the Servicedata from scanned advertisement ? , I am already able to advertise servicedata by editing GAP.cpp from https://github.com/sandeepmistry/ArduinoBLE/commit/55336434a859ccfc830379f56297a9e6f17a6ee0 and it works but still struggling to read the advertised servicedata in Arduino Nano 33 BLE

Polldo, 2020-10-02:
Hi @uasiddiqi , I'm currently working on making advertising and scan response data more flexible. You can take a look here https://github.com/Polldo/ArduinoBLE/commits/advertising_scanresponse_data
It's still a work in progress, so if you find any issues or have any suggestions please tell me.

uasiddiqi, 2020-10-05:
Thank you so much @Polldo  for your quick response. Its nice to hear that someone else is also working in a similar project . I was going through the issues you have worked but its better to ask you here directly. Have you tried reading the servicedata from another device/beacon that is advertising it? . I do realize the effort you have put on scan response but somehow i am unable to relate it to get the servicedata.