arduino-libraries/ArduinoBLE #41: Forced to sleep mode after a while?

alexisicte, 2019-11-06:
Hello,

Here is my sketch:
`#include <ArduinoBLE.h>
#include "LSM9DS1.h"
//Services
BLEService BatteryService("180F");
BLEService SensorService("92a041f8-69de-48ae-a610-6d2e54854fe0");
//Characteristics
BLEUnsignedCharCharacteristic batteryLevelChar("2A19", BLERead | BLENotify);
BLEStringCharacteristic accelerometer_data("abb6f371-c8d4-4c65-9470-140b686698bd", BLERead | BLENotify, 48);
BLEStringCharacteristic gyroscope_data("f1abb263-686f-4a5b-8a8f-39690d3f3020", BLERead | BLENotify, 48);
BLEStringCharacteristic debug_system("f1abb263-686f-4a5b-8a8f-39690d3f3020", BLERead | BLENotify, 48);

String values_to_send = " ";
float x_accel, y_accel, z_accel, x_gyro, y_gyro, z_gyro;
float x_accel_f, y_accel_f, z_accel_f, x_gyro_f, y_gyro_f, z_gyro_f = 0;
String debug_to_send = " ";

void setup() {
  Serial.begin(9600);

  pinMode(LED_PWR, OUTPUT);
  digitalWrite(LED_PWR, LOW);
  
  if (!lsm9ds1.begin()) {
    Serial.println("Failed to initialize lsm9ds1!");
    while (1);
  }
  lsm9ds1.init();// initialize sensors and parameters
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }

  // set advertised local name and service UUID:
  BLE.setLocalName(SENSOR_NAME);
  BLE.setAdvertisedService(BatteryService);
  BLE.setAdvertisedService(SensorService);
  // add the characteristic to the service
  BatteryService.addCharacteristic(batteryLevelChar);
  SensorService.addCharacteristic(accelerometer_data);
  SensorService.addCharacteristic(gyroscope_data);
  SensorService.addCharacteristic(debug_system);
  // add services
  BLE.addService(BatteryService);
  BLE.addService(SensorService);
  // start advertising
  BLE.advertise();
  Serial.println("BLE Peripheral");
}

void loop() {
  // listen for BLE peripherals to connect:
  BLEDevice central = BLE.central();

  // if a central is connected to peripheral:
  if (central) {
    Serial.print("Connected to central: ");Serial.println(central.address());
    Serial.print("Acceleration binary status: "); Serial.println(lsm9ds1.read_accel_status(), BIN);
    Serial.print("Gyroscope binary status: "); Serial.println(lsm9ds1.read_gyro_status(), BIN);
    // print the central's MAC address:
    Serial.println(central.address());

    // while the central is still connected to peripheral:
    while (central.connected()) {
      if (lsm9ds1.accelerationAvailable()) {
        lsm9ds1.readAcceleration(x_accel, y_accel, z_accel);
        // set alpha to zero if you don't want to apply the low pass filter
        accelerometer_data.writeValue(low_pass_filter(x_accel, y_accel, z_accel, 0.01, 0));
      }
      if (lsm9ds1.gyroscopeAvailable()) {
        lsm9ds1.readGyroscope(x_gyro, y_gyro, z_gyro);
        gyroscope_data.writeValue(low_pass_filter(x_gyro, y_gyro, z_gyro, 0.01, 1));
      }
      updateBatteryLevel();
      update_temperature();
    }
    // when the central disconnects, print it out:
    Serial.print("Disconnected from central: ");
    Serial.println(central.address());
  }
  BLE.advertise(); // start advertising again 
}
void updateBatteryLevel() {
  int batteryLevel = map(analogRead(A0), 612, 815, 0, 100);//815 is analog max
//  Serial.print("analogRead(A0): ");Serial.println(analogRead(A0));
//  Serial.print("batteryLevel: ");Serial.println(batteryLevel);
//  delay(250);
  batteryLevelChar.writeValue(batteryLevel);  // and update the battery level characteristic
}
void update_temperature() {
  debug_to_send = " ";
  lsm9ds1.read_system_temperature();
  debug_to_send += String(lsm9ds1.temperature) + "|";
  debug_to_send += String(analogRead(A0));
  debug_system.writeValue(debug_to_send);
}

String low_pass_filter(float x_axis, float y_axis, float z_axis, float alpha, int sensor) {
  values_to_send = " ";// re initialize array
  if (!sensor) { // sensor = accelerometer
    x_accel_f = alpha * x_accel_f + (1 - alpha) * x_axis;
    y_accel_f = alpha * y_accel_f + (1 - alpha) * y_axis;
    z_accel_f = alpha * z_accel_f + (1 - alpha) * z_axis;
    values_to_send += String(x_accel_f) + "|";
    values_to_send += String(y_accel_f) + "|";
    values_to_send += String(z_accel_f);
  }
  else {
    x_gyro_f = alpha * x_gyro_f + (1 - alpha) * x_axis;
    y_gyro_f = alpha * y_gyro_f + (1 - alpha) * y_axis;
    z_gyro_f = alpha * z_gyro_f + (1 - alpha) * z_axis;
    values_to_send += String(int(x_gyro_f)) + "|";
    values_to_send += String(int(y_gyro_f)) + "|";
    values_to_send += String(int(z_gyro_f));
  }
  return values_to_send;
}`

In short the sketch creates services and characteristics and advertises it self. When a central is connected, i write some values on the characteristics. I am using nrf connect android app to debug the procedure. 

Everything running smoothly expects when waiting for a central for "too long". I cannot specify "too long" time period yet, but i am sure is about 20-30 minutes. 
When this time period exceeds, while my device still adverises and seeking for a connection, despite the fact that the device is discoverable and connectable, the central device cannot read its services. More specifically, it seems that the connection is made (i am assuming that because the application let me to disconnect from the device ) and tries to read the advertised services without a success. After a few seconds the connection is lost.

I am assuming that this is not a bug from the application because the same thing happens with other apps too. I believe that the device is going to sleep mode by default (i have to prove that by measuring the current). Despite that, the "library" should interact with radio signal and quit from sleep mode, if this is the case.

Does anyone experienced something similar?
Is it true that the device is forced to sleep mode? and if yes why central's cannot read its services?

P.S. of cource after a reset the device works properly

sandeepmistry, 2019-11-11:
>  but i am connecting 3.3v pin direct to lipo battery (4.2-3.2 V). I dont have a clue that this is the problem.

This is not recommended at all from my knowledge! @pnndra @agdl any thoughts on this?

sandeepmistry, 2019-11-11:
> When the device is attached to usb the above problem never occurs.

Have you tried to remove all the `Serial.print` statements in your sketch?

alexisicte, 2019-11-12:
> > but i am connecting 3.3v pin direct to lipo battery (4.2-3.2 V). I dont have a clue that this is the problem.
> 
> This is not recommended at all from my knowledge! @pnndra @agdl any thoughts on this?

If i am reading the nano 33 ble board layout correctly, the 3.3v trace is "created" from the mpm3610 ic. The output pins from mpm3610 ic are generating the N$16 trace which is by default connected to 3.3v pin. But as i said at the beginning my configuration is "cutted 3.3V jumper". Therefore, i can say that the 3.3v pin is not actually 3.3v but unregulated battery voltage. 

[image]

Please correct me if i am reading in a wrong way the board layout.

Both nina module and lsm9ds1 sensor can be powered up to 3.6v regarding their datasheets. So, i am powering the board with max 4.2 down to 3.2. Therefore, there is 0.6v offset, which is for sure not recommended but i do not believe that this is not the case here.

alexisicte, 2019-11-12:
> > When the device is attached to usb the above problem never occurs.
> 
> Have you tried to remove all the `Serial.print` statements in your sketch?

Yes, i just try without any Serial method. Still the same error: **Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST**

sandeepmistry, 2019-11-14:
@alexisicte ok, so do you think this is a duplicate of issue https://github.com/arduino-libraries/ArduinoBLE/issues/33 now?

pnndra, 2019-11-15:
Hi,
Powering the board directly from battery is not correct. The maximum operating voltage is 3.6v and while this is more or less the voltage a battery will output most of the time, the 4.2v voltage of a fully charged battery WILL damage the board and produce unpredictable results.
Note also that when battery is at or below 3.3v it is basically deleted. This means that drawing current from it will likely damage it in the long term. In any case, when the battery is discharged it won't be able to withstand current spikes and when these happen the voltage will drop significantly. With a multimeter you won't see this as it averages measurements but while Nina usually consumes almost nothing when idle, it will require requires several mA for a very short time and this will likely cause voltage to drop enough to go below allowed voltage.
If you plan to have a very low power application I would recommend using a ldo with low quiescent current so that most of the time it won't consume almost anything (Nina can go down to about 5uA which makes less than 5 uW, not including ldo's iq) it won't impact battery life significantly but will stabilize the board. I would also recommend using a voltage monitor that cuts off battery when voltage drops below 3.3v as this will save your battery from overdischarge.

alexisicte, 2019-11-16:
@pnndra , @sandeepmistry  Many thanks for your input and advices.

closed 2019-11-16 by alexisicte