arduino-libraries/ArduinoBLE #46: Slowly increasing BLE Scan return times

SamoJS, 2019-12-28:
I have a range monitoring application on an Arduino Nano 33 BLE. It monitors for a specific BLE peripheral and will perform an alarm function if that peripheral is not in range.  

The issue I'm seeing is slowly increasing BLE.scanForAdress or BLE.scanForName times. The log entries show the device and details, last field is the elapsed time in ms using millis(); The elapsed time starts out quite useable, the code sleeps 5 seeconds (delay(5000)) after a successful return from BLE.scanForAddress, and 50ms if unsuccessful.  Heres a log example where the scan is returning as expected....

13:57:24.387 -> BLE Scan Initiated
13:57:25.318 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -45 Time: 886
13:57:30.328 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -45 Time: 0
13:57:35.476 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -45 Time: 75
13:57:40.668 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 130
13:57:45.816 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -45 Time: 100
13:57:50.833 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
13:57:55.975 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 75
13:58:01.052 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -41 Time: 0
13:58:06.099 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -41 Time: 0
13:58:11.119 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -39 Time: 0
13:58:16.158 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
13:58:21.296 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 103
13:58:26.330 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -47 Time: 0
13:58:31.353 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
13:58:36.433 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -47 Time: 20
13:58:41.510 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 20
13:58:46.518 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -41 Time: 0
13:58:52.207 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -47 Time: 640
13:58:57.293 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 21
13:59:02.301 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 0
13:59:08.373 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 1000
13:59:13.374 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
13:59:18.409 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 0
13:59:23.439 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
13:59:28.447 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 0
13:59:33.532 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -47 Time: 20
13:59:38.597 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 22
13:59:43.680 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -41 Time: 19
13:59:48.707 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
13:59:53.848 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 72
13:59:58.930 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 26
14:00:03.938 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -49 Time: 0
14:00:08.954 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 0
14:00:13.977 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 0
14:00:19.123 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 73
14:00:24.192 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 21
14:00:29.301 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 73
14:00:34.325 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 0
14:00:39.414 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -46 Time: 22
14:00:44.438 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -42 Time: 0

Slowly the return times increase.... here's an output 4 hours later, eventually it will go out to 8000+ elapsed times.... Not usable.

18:04:40.282 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -45 Time: 19
18:04:45.805 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 482
18:04:52.855 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -40 Time: 1976
18:04:57.968 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -45 Time: 70
18:05:03.056 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -44 Time: 19
18:05:08.534 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 431
18:05:13.596 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 25
18:05:19.146 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 482
18:05:24.675 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -44 Time: 480
18:05:29.745 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 20
18:05:35.268 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -40 Time: 484
18:05:40.355 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -40 Time: 22
18:05:46.427 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 999
18:05:51.950 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -40 Time: 484
18:05:57.025 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -44 Time: 22
18:06:09.546 -> BLE Device - TIMED OUT... Time: 7514
18:06:13.156 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 3081
18:06:20.263 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 2025
18:06:26.816 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 1507
18:06:33.367 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -38 Time: 1511
18:06:39.431 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -40 Time: 991
18:06:45.997 -> BLE Device - Name: BR532397 ADDR: 20:18:ff:00:3f:e2 RSSI: -44 Time: 1508
[... 79 lines cut ...]
        //increment counter
        missCounter++;
      
        if (missCounter >= maxMisses)
        {
          Serial.println("BLE Device - Missing.... ALL STOP!!");
        }  
        else      
        {      
          Serial.print("BLE Device - TIMED OUT... Time: ");
          Serial.println(sTime-eTime);
        }

        //reset the elapsed time counter
        eTime = millis();
        sTime = eTime;

      }
        
      // wait for a bit  
      delay(50);
 
  }
  
}

SamoJS, 2019-12-28:
Issue was local interference on the RF spectrum....

closed 2019-12-28 by SamoJS

remijaboeuf, 2021-12-15:
hello!
I face exactly the same issue with a nano 33 iot.
Could you find the reason for this increasing connection delay? or at least a way to bypass it?
I tried several things such as stoping scan and restarting the scan periodically, but it didn't prove so successful.
thank you so much in advance!!