arduino-libraries/ArduinoBLE #47: Support Adafruit feather express nrf52840

marcrupprath, 2019-12-29:
Hello,
it would be nice if you can support Adafruit nrf52840 express feater, too

Marc

orhanyor, 2020-01-03:
> Hello,
> it would be nice if you can support Adafruit nrf52840 express feater, too
> 
> Marc

what is the difference in adafruit boards? considering they have the identical nrf chip this library should work in their boards too.

marcrupprath, 2020-01-04:
Hello,
i first test lib with example sketch "scanning" and Ardoino 33 ano BLE, all as expcted
I then connected Adafruit nrf52840 feather express and got following eror message:

 #error "Unsupported board selected!"

Is there a header file where i can make the needed adjustments and where to find ?

Thanks
Marc