arduino-libraries/ArduinoBLE #48: Sending classes

skelstar, 2019-12-30:
Trying to send my own object from peripheral to central and from what I can work out, this should work:

_peripheral_:
```
class TestClass
{
  public:
    unsigned long id;
};

TestClass t;

void sendData()
{
  testDataChar.writeValue(&t, sizeof(TestClass)); // and update the battery level characteristic

  t.id = t.id + 1;
  Serial.printf("id: %lu\n", t.id);
}
```

_central_:
```
class TestClass
{
  public:
    unsigned long id;
};

void readData()
{
    since_read_client = 0;
    TestClass t;
    characteristic.readValue(&t, sizeof(TestClass));
    Serial.printf("id: %lu\n", t.id);
}
```

... but all I'm getting back is `id: 0` for each packet sent. Is there something I'm doing wrong? I couldn't find any examples anywhere.

joonazan, 2020-02-05:
You're probably doing something wrong in initializing the characteristic.

I am using an undocumented `BLETypedCharacteristic`, which has a template parameter.

```
BLETypedCharacteristic<FieldStrength> field_strength_char(
  "c478c8cc-1287-4b01-b503-87399d9d835f",
  BLERead | BLENotify
);
```

thekunalsaini, 2020-03-30:
Hi @skelstar,
All looking fine but one I will suggest is that please check parameters characteristic.readValue(&t, sizeof(TestClass));  this function should accept.
Hope it helps you