arduino-libraries/ArduinoBLE #50: Communication of One  Central Devicel and One  Peripheral Devices

vyohai, 2020-01-12:
Does the library support connection and communication between one  central device and more than one peripheral device?
And how could it be implemented?

Polldo, 2020-08-11:
Hi @vyohai ,
yes, it is possible. 
Each central device can be connected to more peripherals and each peripheral can be connected to more centrals. 
However, at the moment the library doesn't expose the functions to retrieve the connected devices, so the handling of multiple devices must be done at application level. 
To deal with this I've implemented some possible functions that allow to loop through all the connected devices. You can see it here https://github.com/arduino-libraries/ArduinoBLE/pull/105 , let me know what you think could be added or improved.
Here are two examples of how to use these additional functions to manage multiple connected devices.

--- Peripheral device handling more centrals:
``` C++
#include <ArduinoBLE.h>

BLEService batteryService("180F");
BLEUnsignedCharCharacteristic batteryLevelChar("2A19", BLERead | BLEWrite); 

void setup() {
  Serial.begin(115200);    
  pinMode(LED_BUILTIN, OUTPUT); 
  // Start BLE
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }
  BLE.debug(Serial);

  // Configure peripheral
  BLE.setLocalName("peripheral");
  BLE.setAdvertisedService(batteryService); 
  batteryService.addCharacteristic(batteryLevelChar); 
  BLE.addService(batteryService);
  BLE.advertise();

  Serial.println("peripheral configured - central connected: 0");
}

void loop() {
  BLE.poll();
  // Led on if some central is connected
  if (BLE.centralCount() > 0) {
    digitalWrite(LED_BUILTIN, HIGH);
  } else {
    digitalWrite(LED_BUILTIN, LOW);
  }

  // Check number of connected devices and enable advertising
  static auto timeRef = millis();
  if (millis() - timeRef >= 5000) {
    timeRef = millis();
    auto count = BLE.centralCount();
    Serial.print("Central count: ");
    Serial.println(count);
    if (count < 3) {
      BLE.advertise();
    }
  }
}
```
--- Central device handling more peripherals:
``` C++
#include <ArduinoBLE.h>

void setup() {
  Serial.begin(115200);
  BLE.begin();
  BLE.scanForName("peripheral");
}

void connectionLoop() 
{
  // If not already connected to 2 peripherals, try to connect to a new found peripheral.
  if (BLE.peripheralCount() < 2) {

    BLEDevice peripheral = BLE.available();
    if (peripheral) {
      BLE.stopScan();

      if (!peripheral.connect()) {
        Serial.println("Failed to connect!");
        return;
      }

      if (!peripheral.discoverAttributes()) {
        Serial.println("Attribute discovery failed!");
        peripheral.disconnect();
        return;
      }
    }
  }
}

auto timeRef = millis();
void loop() {
  connectionLoop();

  if (millis() - timeRef >= 5000) {
    timeRef = millis();

    int periphCount = BLE.peripheralCount();
    Serial.print(" Peripheral connected: ");
    Serial.println(periphCount);

    if (periphCount < 2) {
      BLE.scanForName("peripheral");
    }

    // Loop through all connected peripherals
    for (int periphIdx = 0; periphIdx < periphCount; periphIdx++) {
      BLEDevice peripheral = BLE.peripheral(periphIdx);
      if (peripheral) {
        BLECharacteristic batteryLevelChar = peripheral.characteristic("2A19");

        if (!batteryLevelChar) {
          Serial.println("Peripheral does not have battery level characteristic!");
          peripheral.disconnect();
        } else {
          Serial.print("Peripheral connected, value: ");
          batteryLevelChar.read();
          Serial.println(*batteryLevelChar.value());
        }
      }
    }
  }
}
```

vyohai, 2020-08-18:
Hi Polldo
thank you for all of your work
the example code doesn't work.
the error is :
'class BLELocalDevice' has no member named 'peripheralCount'
and in general when I try to scan for 2 peripherals and then connect to them one by one the 2nd one can't connect.
could that work?
am I doing something wrong?

Polldo, 2020-08-18:
@vyohai to compile the example you should checkout the `multi-connection` branch: https://github.com/arduino-libraries/ArduinoBLE/tree/multi-connection
Also, are you using a nano 33 ble? In that case there is an open issue that addresses the problem of connecting a central to many peripherals https://github.com/arduino-libraries/ArduinoBLE/issues/108

vyohai, 2020-08-21:
thank you it seems like it's working

closed 2020-08-21 by vyohai

sprasadkrish, 2020-09-08:
> Hi Polldo
> thank you for all of your work
> the example code doesn't work.
> the error is :
> 'class BLELocalDevice' has no member named 'peripheralCount'
> and in general when I try to scan for 2 [...]

> 
> 
> thank you it seems like it's working

Hi @vyohai 
           how you solved this issue , i am also facing same kind of issue .

Polldo, 2020-09-08:
Hi @sprasadkrish ,
can you describe the issue you are facing?
Have you tried this https://github.com/arduino-libraries/ArduinoBLE/issues/50#issuecomment-672024611 ?
Also, if you are using a nano33ble have a look at this https://github.com/arduino-libraries/ArduinoBLE/issues/108
Thanks

vyohai, 2020-09-08:
this is part of my code that runs on the master, hope you will find it halpfull
`
if (periphCount < 2)
        {
          connecting("per1", 0);
          connecting("per2", 1);
          BLE.stopScan();
        }

void connecting(String namee, int num)
{

  //check if peripheral1 connected
  BLEDevice p0 = BLE.peripheral(num);
  //loop until peripheral1 is connected
  BLE.scanForName(namee);
  BLEDevice a0 = BLE.available();
  while (!a0.connected())
  {
    if (Debug)
      Serial.println("scaning...");
    while (!a0)
    {
      a0 = BLE.available();
    }

    if (a0.connect())
    {
      if (Debug)
      {
        Serial.print("conected");
        Serial.println(num + 1);
      }
    }
    else
    {
      if (Debug)
      {
        Serial.print("failed to connect");
        Serial.println(num + 1);
      }
    }
  }
}
`

and than when you want to use them:
`int periphCount = BLE.peripheralCount();
BLEDevice p0 = BLE.peripheral(0);
initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);
BLEDevice p1 = BLE.peripheral(1);
BLECharacteristic valC1 , tsC1;
initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);
`

sprasadkrish, 2020-09-08:
> 
> 
> Hi @sprasadkrish ,
> can you describe the issue you are facing?
> Have you tried this [#50 (comment)](https://github.com/arduino-libraries/ArduinoBLE/issues/50#issuecomment-672024611) ?
> Also, if you are using a nano33ble have a look at this #108
> Thanks

Hi @vyohai 
                 Thank you , my issue : MKR1010 BLE - Timeout issue #109 
[#109 ](url)  
                     https://github.com/arduino-libraries/ArduinoBLE/issues/109

vyohai, 2020-09-08:
this is part of my code that runs on the master, hope you will find it halpfull
```if (periphCount < 2)
{
     connecting("per1", 0);
     connecting("per2", 1);
     BLE.stopScan();
}`
`void connecting(String namee, int num)
{
  //check if peripheral1 connected
  BLEDevice p0 = BLE.peripheral(num);
  //loop until peripheral1 is connected
  BLE.scanForName(namee);
  BLEDevice a0 = BLE.available();
  while (!a0.connected())
  {
    if (Debug)
      Serial.println("scaning...");
    while (!a0)
    {
      a0 = BLE.available();
    }
    if (a0.connect())
    {
      if (Debug)
      {
        Serial.print("conected");
        Serial.println(num + 1);
      }
    }
    else
    {
      if (Debug)
      {
        Serial.print("failed to connect");
        Serial.println(num + 1);
      }
    }
  }
}
```

and than when you want to use it
```
int periphCount = BLE.peripheralCount();

BLEDevice p0 = BLE.peripheral(0);

initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);

BLEDevice p1 = BLE.peripheral(1);

BLECharacteristic valC1 , tsC1;

initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);

void initChar(BLEDevice &p0, const char* Char1name, const char* Char2name, const char* Char3name, BLECharacteristic &valC0, BLECharacteristic &tsC0, BLECharacteristic &tmC0)
{
p0.discoverAttributes();
valC0 = p0.characteristic(Char1name);
tsC0 = p0.characteristic(Char2name);
tmC0 = p0.characteristic(Char3name);
}
```

sprasadkrish, 2020-09-08:
> 
> 
> this is part of my code that runs on the master, hope you will find it halpfull
> 
> ```
> {
>      connecting("per1", 0);
>      connecting("per2", 1);
>      BLE.stopScan();
> }`
> `void connecting(String namee, int num)
> {
>   //check if peripheral1 connected
>   BLEDevice p0 = BLE.peripheral(num);
>   //loop until peripheral1 is connected
>   BLE.scanForName(namee);
>   BLEDevice a0 = BLE.available();
>   while (!a0.connected())
>   {
>     if (Debug)
>       Serial.println("scaning...");
>     while (!a0)
>     {
>       a0 = BLE.available();
>     }
>     if (a0.connect())
>     {
>       if (Debug)
>       {
>         Serial.print("conected");
>         Serial.println(num + 1);
>       }
>     }
>     else
>     {
>       if (Debug)
>       {
>         Serial.print("failed to connect");
>         Serial.println(num + 1);
>       }
>     }
>   }
> }
> ```
> 
> and than when you want to use it
> 
> ```
> int periphCount = BLE.peripheralCount();
> 
> BLEDevice p0 = BLE.peripheral(0);
> 
> initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);
> 
> BLEDevice p1 = BLE.peripheral(1);
> 
> BLECharacteristic valC1 , tsC1;
> 
> initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);
> 
> void initChar(BLEDevice &p0, const char* Char1name, const char* Char2name, const char* Char3name, BLECharacteristic &valC0, BLECharacteristic &tsC0, BLECharacteristic &tmC0)
> {
> p0.discoverAttributes();
> valC0 = p0.characteristic(Char1name);
> tsC0 = p0.characteristic(Char2name);
> tmC0 = p0.characteristic(Char3name);
> }
> ```

Hi @vyohai ,
                     Thanks a lot , i will try this and update you.

sprasadkrish, 2020-09-09:
Hi @vyohai 
                       i am getting this error  'Debug' was not declared in this scope ,
if i disable Debug , i am unable to connect any peripherals .

vyohai, 2020-09-09:
I will upload the whole master code:

```
#include <ArduinoBLE.h>
#include <String>
//config
//0-timing 1 slave 1 master communication
//1-1 slave 1 master send to matlab
//2-2 slaves 1 master send to matlab
//3-2 slaves 1 master send to matlab 2 way communication
bool Debug = HIGH;
byte numSlaves = 3;
int WHAT=0;
//functions
void connecting(String, int);
void initChar(BLEDevice & , const char* , const char*, const char* , BLECharacteristic &, BLECharacteristic &, BLECharacteristic &);
void sendData(BLECharacteristic, BLECharacteristic,bool,int,int);
void sendMasterData(bool Debug);
uint16_t sync(BLECharacteristic ,BLECharacteristic ,bool ,int );
void setup() {
  //setup serial
  Serial.begin(115200);
  //  setup ble
  BLE.begin();
  //BLE.debug(Serial);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
}
void loop() {
  int periphCount = BLE.peripheralCount();
  BLEDevice p0 = BLE.peripheral(0);
  BLECharacteristic valC0 , tssC0 , tsmC0;//s means slave m means master
  switch (numSlaves) {
    case 0:
      {
        if (periphCount < 1)
        {
          connecting("per1", 0);
          BLE.stopScan();
        }
        //define charachtiristics
        initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);
        //while the slaves connected send data
        while (p0.connected() )
        {
          uint32_t t0 = millis();
          uint16_t ts, a;
          valC0.readValue(a);
          tssC0.readValue(ts);
          Serial.print( "dt is ");
          Serial.println(millis() - t0);
          digitalWrite(LED_BUILTIN, HIGH);
        }
        break;
      }
    case 1:
      {
        if (periphCount < 1)
        {
          connecting("per1", 0);
          BLE.stopScan();
        }
        //define charachtiristics
       initChar( p0, "0101", "0102","0103", valC0, tssC0, tsmC0);
        //while the slaves connected send data
        while (p0.connected() )
        {
          uint32_t t0 = millis();
          sendData(tssC0, valC0, Debug, 0,numSlaves);
          Serial.print( "dt is ");
[... 204 lines cut ...]
  tssC0.readValue(ts2);
  tm2=millis();
  dt=tm1/2-ts2+tm2/2;
  if (Debug)
  {
    Serial.print("hello from sync");
    Serial.println(slave);
    Serial.print("tm1 is ");
    Serial.println(tm1);
    Serial.print("tm2 is ");
    Serial.println(tm2);
    Serial.print("ts2 is ");
    Serial.println(ts2);
    Serial.print("bye from sync");
    Serial.println(slave);
  }
  else
  {
    Serial.write(dt);
    Serial.write(dt >> 8);
    Serial.write(dt >> 16);
    Serial.write(dt >> 24); 
  }
}
```

sprasadkrish, 2020-09-09:
Hi @vyohai 
                Thank you , by using this code , i will do changes in my code.

Prashant123-lab, 2021-06-14:
> I will upload the whole master code:
>
> ```
> #include <ArduinoBLE.h>
> #include <String>
> //config
> //0-timing 1 slave 1 master communication
> //1-1 slave 1 master send to matlab
> //2-2 slaves 1 master send [...]

Hey @vyohai 
How did you resolve this issue 'class BLELocalDevice' has no member named 'peripheralCount'?