arduino-libraries/ArduinoBLE #54: Recompile times very long

tikonen, 2020-01-24:
I'm not sure if this is problem of ArduinoBLE library or ArduinoIDE but looks like the IDE recompiles whole ArduinoBLE library every time, even when nothing has changed in the user project code or settings. This makes code iteration very slow, especially because Arduino build environment does not support multicore building.

Here is verbose build log for reference.

`C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Teemu Ikonen\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -vid-pid=0X2341_0X805A -ide-version=10810 -build-path C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_build_47769 -warnings=default -build-cache C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_cache_195006 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\Teemu Ikonen\Documents\Arduino\nano33_ble_test\nano33_ble_test.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Teemu Ikonen\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -vid-pid=0X2341_0X805A -ide-version=10810 -build-path C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_build_47769 -warnings=default -build-cache C:\Users\TEEMUI~1\AppData\Local\Temp\arduino_cache_195006 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\Teemu Ikonen\Documents\Arduino\nano33_ble_test\nano33_ble_test.ino
Using board 'nano33ble' from platform in folder: C:\Users\Teemu
Using core 'arduino' from platform in folder: C:\Users\Teemu
Detecting libraries used...
"C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\TEEMUI~1\\AppData\\Local\\Temp\\arduino_build_47769\\sketch\\nano33_ble_test.ino.cpp" -o nul
Alternatives for ArduinoBLE.h: [ArduinoBLE@1.1.2]
ResolveLibrary(ArduinoBLE.h)
  -> candidates: [ArduinoBLE@1.1.2]
"C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Teemu Ikonen\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\TEEMUI~1\\AppData\\Local\\Temp\\arduino_build_47769\\sketch\\nano33_ble_test.ino.cpp" -o nul
"C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\Teemu Ikonen\\Documents\\Arduino\\libraries\\ArduinoBLE\\src" "-IC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\cores\\arduino" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\Teemu Ikonen\\Documents\\Arduino\\libraries\\ArduinoBLE\\src\\BLECharacteristic.cpp" -o nul
"C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.4\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\Teemu [... cut ...]
Using library ArduinoBLE at version 1.1.2 in folder: C:\Users\Teemu Ikonen\Documents\Arduino\libraries\ArduinoBLE
"C:\\Users\\Teemu Ikonen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-size" -A "C:\\Users\\TEEMUI~1\\AppData\\Local\\Temp\\arduino_build_47769/nano33_ble_test.ino.elf"
Sketch uses 284228 bytes (28%) of program storage space. Maximum is 983040 bytes.
Global variables use 65664 bytes (25%) of dynamic memory, leaving 196480 bytes for local variables. Maximum is 262144 bytes.
`

alexisicte, 2020-01-24:
I had similar issues. I believe the Arduino IDE does this. The library has nothing to do with that. Compiling on platform.io - VS code the compilation is really fast on all cases.

facchinm, 2020-01-24:
Hi @tikonen ,
it is a known issue about library dependencies that regressed in 1.8.10. Latest hourly (and `arduino-builder` > 1.4.5) contains the fix, and will be shipped very soon as part of IDE 1.8.11.

closed 2020-01-24 by facchinm