arduino-libraries/ArduinoBLE #55: Multiple concurrent connections

javieragst, 2020-01-31:
Hi everyone,
I'm trying to connect 3 ble modules working as peripherals to an Arduino Nano 33 BLE working as central. In the datasheet of the micro says that it can handle 20 bluetooth connections, but I can't connect to a second peripheral when it is one connected already. Taking a quick view to the library I have not found any problem to connect more than one device, I only found this parameter ATT_MAX_PEERS that I understand that limit the connections to three. It is my first time using BLE  and I'm not agood programmer, so I don't know if my reasoning is correct.
It is possible to connect more than one peripheral to the Arduino Nano 33 BLE using this library?
Thanks

aravitz1, 2020-05-24:
I would really like to know the answer to this. I’m attempting to do the same thing but cannot figure out how to implement more than one connection from a single central to multiple peripherals.

Gerriko, 2020-06-04:
Another scenario I am looking at is to get 2 central devices (e.g. phones) to connect to a single BLE peripheral using the Arduino nano 33 ble board.

Reference: https://devzone.nordicsemi.com/f/nordic-q-a/24160/multiple-central-to-single-peripheral-connections

gnsmrky, 2020-08-10:
@javieragst @aravitz1 Have you found out a way around this problem?  Also facing the same situation.

Polldo, 2020-08-17:
Hi @javieragst @aravitz1 @Gerriko @gnsmrky ,
as reported in this issue https://github.com/arduino-libraries/ArduinoBLE/issues/108, latest versions of the nano33ble core have problems with multiple connections. Follow that issue to receive updates.
Regarding https://github.com/arduino-libraries/ArduinoBLE/issues/55#issuecomment-638812933, one peripheral connected to more centrals should work even if using latest nano33ble core versions. Refer to https://github.com/arduino-libraries/ArduinoBLE/issues/50 for a working example.

closed 2020-08-17 by Polldo