arduino-libraries/ArduinoBLE #57: `BLE.poll` waits for next data transfer event.

joonazan, 2020-02-05:
This can be found out by experimenting with different connectionIntervals and measuring the maximum time that `BLE.poll` takes.

This behaviour is very inconvenient, because it wastes a lot of CPU cycles on MKR Wifi 1010. The only workaround would be to do all computation in interrupts.

Polldo, 2020-07-13:
Hi @joonazan ,
the poll function performs a loop until all the received packets are analyzed. If no packet (or event) is available, the function returns immediately. 
So, there is no wait for next data reception. 
Moreover, the poll function could be called with a timeout parameter. In such case, the function will wait for an event until the timeout is over.

joonazan, 2020-07-13:
@Polldo In my measurements the poll always took exactly as long as the connection interval that I set. I was sending data all the time, so it is probably related to that.

I am no longer developing the project where this was a problem but I'd assume that if the problem still exists it will bother anyone trying get a measurement over I²C every millisecond and sending a notification about them from time to time.

My project ended up using [my own library](https://github.com/joonazan/nina-fast-bluetooth) instead.

Polldo, 2020-07-13:
@joonazan 
>I was sending data all the time, so it is probably related to that.

Yes, that should be the cause.

Anyway cool project, thanks for sharing.

closed 2020-07-13 by Polldo