arduino-libraries/ArduinoBLE #58: setManufacturerData not advertising for me w/MKR WIFI1010 brd

zumbrun, 2020-02-18:
Hi, despite setting the ble manufacturer data in code, it is not advertised based on 4-5 different ble peripheral detector apps including Light Blue.  The local name, device name, and service id are being advertised...just the manufacturer data is never visible.

Tried smaller byte sizes of manufacturers data...I think we are well under the max byte size for an advertising packet. Tried to advertise just manufacturer data w/o service id, tried to advertise w/ the optional argument of company id....nothing seems to make it visible.  Since other information set in code is being advertised, then it seems like something specific to manufacturer data is an issue.  The lib source code seems good but I'm not an expert.  Hope you can help us!  We need this for our product.

Here is latest code snippet.
while (!Serial);
  //begin initialization
  if (!BLE.begin()) {
    Serial.println("starting BLE failed!");
    while (1);
  }
  Serial.println("Bluetooth device active, waiting for connections...");
  // connect to HG iphone apps
  BLE.setLocalName("ICBHR");
  BLE.setDeviceName("ICBHR2");
  //
  uint8_t data[10] = {0x31, 0x37, 0x30, 0x33, 0x30, 0x30, 0x32, 0x37, 0x31, 0x31};
  BLE.setManufacturerData(data, sizeof(data));
  BLE.setAdvertisedService(HGService);
  BLE.addService(HGService);
  BLE.advertise();
}

zumbrun, 2020-02-18:
p.s. posted on forum.  there were about 30 views but no responses.

thekunalsaini, 2020-03-20:
@zumbrun What issue you are facing here?

zumbrun, 2020-03-20:
The issue for us is when setManufacturerData is used, it doesn't show up in advertising packet.  It seems like it should based on docs.  (setLocalName and setDeviceName do show up.)

fgaetani, 2020-03-23:
Hi @zumbrun, I have found the same issue on Arduino Nano 33 BLE. Have you solved somehow? Thank you

facchinm, 2020-03-23:
Closing as duplicate of #56 , a fix is coming

closed 2020-03-23 by facchinm