arduino-libraries/ArduinoBLE #59: How to activate Bootloader? (Nano 33 BLE - for OTA app)

gbucas, 2020-02-25:
Hi guys,

Is there a way to activate the bootloader from code in sketch on the nano 33 BLE?

I want to add an OTA software update function to my iOS App (that already communicates well with the nano 33 BLE). Are there any examples/tips to help me to develop this (OTA) further?

BTW: Thanks for the new StringCharacteristic - makes it easy to emulate the Nordic UART Characteristic (and do many other useful things!).

Thanks!

Gerard

ChanceCuddeback, 2021-03-26:
Hello, I'm attempting to do the same. Did you find a nice resource? So far I'm going to purchase a J-Link and put a new bootloader on the nano 33 BLE via the SWD interface. Then attempt to program the nano with Nordic's software suit. 

Thank you!

gbucas, 2021-07-14:
Still looking for an OTA solution for nano BLE 33 (using Bluetooth). 

Has anyone managed to do this?

thanks

Gerard

ChanceCuddeback, 2021-07-14:
I’m also still looking, I have begun learning Zephyr as it has FOTA capabilities. But I’m not sure if I have to change the boot loader yet (still learning). 
But learning an entire RTOS and build system, seems like a lot just to get over the air updates.

vovagorodok, 2024-04-19:
I'm working on such solution: https://github.com/vovagorodok/ArduinoBleOTA