arduino-libraries/ArduinoBLE #60: #error "Unsupported board selected!"

sajadghawami, 2020-03-11:
Hey there,

i got a `KeeYees ESP32 ESP-32S NodeMCU` with a `ESP-WROOM-32 Chip`.

The problem is, i am currently just trying to import the necessary library to start with BLE:

`
#include <ArduinoBLE.h>
`

But i keep getting:

`/Users/USER/.platformio/lib/ArduinoBLE_ID5878/src/utility/HCIUartTransport.cpp:29:2: error: #error "Unsupported board selected!"
 #error "Unsupported board selected!"`

This is my current `plattformio.ini`:

`
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
`

I could not really find anything. Is it true that the Board is not supported for the standard ArduinoBLE library or am i doing something wrong?

facchinm, 2020-03-11:
Indeed, ESP32 is not supported by ArduinoBLE . Please refer to the right channels (ESP32 repo or platform.io) for support, thanks

closed 2020-03-11 by facchinm

sajadghawami, 2020-03-11:
@facchinm 

wow! Thank you for the quick response!

That still sucks, since i am pretty new to the whole arduino world, and it would've been nice to be able to use the "standard" libraries.

Could you be so kind and name any board that would be considered "standard" where i can use the `ArduinoBLE` library?

facchinm, 2020-03-11:
Sure, it supports Arduino MKR WiFi 1010, Arduino UNO WiFi Rev.2, Arduino Nano 33 IoT, and Arduino Nano 33 BLE

sajadghawami, 2020-03-11:
Thank you very much