arduino-libraries/ArduinoBLE #62: central.address() function returns wrong value

fgaetani, 2020-03-23:
Hi, I noticed this issure on Arduino Nano 33 BLE board, the central.address() function returns an incorrect MAC address. In addition, the value also not appears to be unique and change despite being reconnected to the same central device. I await news for the solution.

facchinm, 2020-03-23:
Hi @fgaetani ,
are you using a phone as central? This could be an address randomizer feature of recent mobile operating systems. Do you have the chance to try calling `central.address()` while connecting to another nano?

fgaetani, 2020-03-23:
Hi @facchinm ,
yes, I use a smartphone, I have tried both Android and iOS obtaining the same result, I did not know this randomization technique. Is it therefore not possible to obtain the real physical address?

Nailer, 2020-04-30:
> Hi @facchinm ,
> yes, I use a smartphone, I have tried both Android and iOS obtaining the same result, I did not know this randomization technique. Is it therefore not possible to obtain the real physical address?

He is correct. Due to privacy reasons, devices no longer send a fixed address that you can store. This can be toggled on newer Android phones, but is always turned on in iOS.

closed 2020-05-01 by facchinm