arduino-libraries/ArduinoBLE #65: begin() doesn't work after end() on Nano BLE 33, have to power cycle

rmlearney-digicatapult, 2020-04-07:
Hi,

The begin() function does not behave as expected following an end() and instead I have to power cycle. This is similar to #33

I've created two minimal examples for people to try below:

Example 1:

1. Initialise BLE
2. Call BLE.disconnect() 10 seconds after connected to central
3. Re-advertise 15 seconds later to allow reconnection

Expected behaviour - disconnect from central and allow reconnection after re-advertising
Actual behaviour - works as expected

```c++
#include <ArduinoBLE.h>

BLEService TestService("DEAD");
BLECharacteristic TestData("BEEF", BLEIndicate | BLENotify, 2, true);
BLEDescriptor TestDescriptor("BEEF", "Test");

uint32_t genericTimer = 0;
bool wasConnected = false;

void setup(){
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);

    initBLE();
}

void loop(){
    BLEDevice central = BLE.central(); // begin listening for centrals to connect

    if(central){
        genericTimer = millis();
        while(central.connected()){
            if(millis() - genericTimer >= 10000){ // Wait 10 seconds after connect
                BLE.disconnect(); // Disconnect BLE
                wasConnected = true;
            }
        }
    }

    if(wasConnected){
        genericTimer = millis();
        while(millis() - genericTimer <= 15000){} // Wait 15 seconds after disconnect
        wasConnected = false;
        BLE.advertise();
    }
}

void initBLE(void){
    BLE.begin();
    BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
    BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);
    BLE.setLocalName("TestName");
    BLE.setDeviceName("Test Device Name");
    TestService.addCharacteristic(TestData);
    TestData.addDescriptor(TestDescriptor);
    BLE.addService(TestService);
    BLE.setAdvertisedService(TestService);
    BLE.advertise();
}
```

Example 2:

1. Initialise BLE
2. Disconnect and end() 10 after seconds of connection to central
3. Wait 15 seconds and try to re-start BLE system again

Expected behaviour - system disconnects, then ends BLE service, and restarts it correctly
Actual behaviour - unable to restart BLE with begin() after calling end()

```c++
#include <ArduinoBLE.h>

BLEService TestService("DEAD");
BLECharacteristic TestData("BEEF", BLEIndicate | BLENotify, 2, true);
BLEDescriptor TestDescriptor("BEEF", "Test");

uint32_t genericTimer = 0;
bool wasConnected = false;

void setup(){

pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, LOW);

initBLE();
}

void loop(){
    BLEDevice central = BLE.central(); // begin listening for centrals to connect

    if(central){
        genericTimer = millis();
        while(central.connected()){
            if(millis() - genericTimer >= 10000){ // Wait 10 seconds after connect
                BLE.disconnect(); // Disconnect BLE
                BLE.end(); // End BLE service
                wasConnected = true;
            }
        }
    }

    if(wasConnected){
        genericTimer = millis();
        while(millis() - genericTimer <= 15000){} // Wait 15 seconds after disconnect
        wasConnected = false;
        initBLE();
    }
}

void initBLE(void){
    BLE.begin();
    BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
    BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);
    BLE.setLocalName("TestName");
    BLE.setDeviceName("Test Device Name");
    TestService.addCharacteristic(TestData);
    TestData.addDescriptor(TestDescriptor);
    BLE.addService(TestService);
    BLE.setAdvertisedService(TestService);
    BLE.advertise();
}
```

rmlearney-digicatapult, 2020-04-08:
Probing `BLELocalDevice::begin()` in `BLELocalDevice.cpp` shows that `HCI.reset()` is the source of the failure to restart after end(). Continuing to investigate. Help welcome.

rmlearney-digicatapult, 2020-04-08:
Continuing to probe. `HCIClass::poll` is failing to enter `while (HCITransport.available())`

Looks like UART is not available after end(), and isn't getting reset for some reason. Didn't think the NANO 33 BLE used UART communication with the BLE module.

rmlearney-digicatapult, 2020-04-08:
For those who want to test this, use the following sketch:

```c++
#include <ArduinoBLE.h>

BLEService TestService("DEAD");
BLECharacteristic TestData("BEEF", BLEIndicate | BLENotify, 2, true);
BLEDescriptor TestDescriptor("BEEF", "Test");

uint32_t genericTimer = 0;
bool wasConnected = false;

void setup(){

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  initBLE();

  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  digitalWrite(LED_BUILTIN, HIGH);
}

void loop(){
  BLEDevice central = BLE.central(); // begin listening for centrals to connect

  if(central){
    genericTimer = millis();
    while(central.connected()){
      if(millis() - genericTimer >= 5000){ // Wait 5 seconds
        BLE.disconnect(); // Disconnect BLE - LED should go out
        BLE.end();
        digitalWrite(LED_BUILTIN, LOW); // Doesn't go out - have to do it manually
        wasConnected = true;
      }
    }
  }

  if(wasConnected){
    genericTimer = millis();
    while(millis() - genericTimer <= 5000){} // Wait 5 seconds
      wasConnected = false;
      initBLE();
  }
}

void initBLE(void){ 
  if(BLE.begin()){
    digitalWrite(LED_BUILTIN, HIGH);
  }
  BLE.setLocalName("TestName");
  BLE.setDeviceName("Test Device Name");
  TestService.addCharacteristic(TestData);
  TestData.addDescriptor(TestDescriptor);
  BLE.addService(TestService);
  BLE.setAdvertisedService(TestService);
  BLE.advertise();
}
```
And insert `digitalWrite(LED_BUILTIN, HIGH);` at `line 81` in `BLELocalDevice.cpp`

This will show the LED re-lighting after end().

But moving that `digitalWrite(LED_BUILTIN, HIGH);` to `line 86` will fail, showing that is the (first at least) code block that fails the second time round.

rmlearney-digicatapult, 2020-04-09:
I have a funny feeling it's because `_rxBuf` is getting deallocated in `end()` but isn't getting reallocated the second time `begin()` is called because the whole class isn't reinstantiated, so the `available()` call fails. Please correct me if I'm wrong.

drewbus-drewbus, 2020-05-09:
I'm experiencing the same bug.

navrit, 2020-05-12:
I'm also experiencing this bug, also using the Nano 33 BLE Sense and with the [latest commit](https://github.com/arduino-libraries/ArduinoBLE/commit/8dc4290ab00b7de8164489768dccf6271f919cc3).

@rmlearney-digicatapult & @drewbus-drewbus, I assume you are also trying to turn the radio on and off at will.

I was looking at the [NINA B306 schematics](https://content.arduino.cc/assets/NANO33BLE_V2.0_sch.pdf), it shows a hardware reset pin (RESET_N) in C3/C4 from the NINA at pin 19 (I guess this means a NINA GPIO?) to P0.18 (is this an Arduino pin?). It's not clear to me whether this input is accessible to us currently from the [U-blox NINA-B3 Series datasheet](https://www.u-blox.com/sites/default/files/NINA-B3_SIM_%28UBX-17056748%29.pdf), see 1.5.1. 

@per1234 @facchinm Could you provide assistance? Turning the radio off and on would be very useful.

Let me know if I can help with testing!

sastorer-ms, 2020-05-21:
I am trying to write a reset function that turns BLE off and then back on again. I'm being blocked by this bug as well!

Let me know if there's anything I can do to assist @rmlearney-digicatapult

rmlearney-digicatapult, 2020-05-21:
@sastorer-ms I'm definitely no expert when it comes to C++ but if you could possibly poke around and confirm my findings that could really help.

Unfortunately I don't know how to raise the priority of this issue with the Arduino team but would love to understand if this is on their backlog.

fgaetani, 2020-07-01:
I have the same bug, what can be the solution?

rmlearney-digicatapult, 2020-07-01:
Hi @fgaetani I'm afraid I don't know. I think we need input from one of the Arduino team like @facchinm or @per1234

facchinm, 2020-07-02:
We are actively investigating all the reports and will post a response later today :slightly_smiling_face:

facchinm, 2020-07-02:
The problem here is that the bleLoop thread is getting terminated on `end()` https://github.com/arduino-libraries/ArduinoBLE/blob/693268ddbceaee0c60dafab806560f59087ac940/src/utility/HCICordioTransport.cpp#L200-L202
The CordioHCIHook `terminate()`, however, doesn't shut down the radio, so calling `BLE.end()` right now will not have the expected behaviour.

Moreover, starting a thread twice is forbidden in mbed (https://os.mbed.com/docs/mbed-os/v6.0/mbed-os-api-doxy/classrtos_1_1_thread.html#ad315c4f13f31fffde51040cbf00518cb) so the second `begin()` will fail anyway.

I'm going to post a patch for the thread start/stop but the only way to have a real `end()` functionality is to patch Cordio sources (and could take longer).

@Polldo

rmlearney-digicatapult, 2020-07-02:
@facchinm thank you! Would it make sense to create a new functionality to separate `.end()` from `.off()`

`.end()` would terminate the loop but leave the radio on, and could be restarted by `.begin()`

`.off()` would terminate the loop and turn the radio off, and could also be restarted by `.begin()`

facchinm, 2020-07-02:
The scope of this library is to abstract the underlying hardware as much as possible, thus a proper `end()` should be implemented in cordio, no need to add other useless APIs :slightly_smiling_face: 

Anyway, I added https://github.com/arduino-libraries/ArduinoBLE/pull/92 to fix the end() -> begin() hang, pleas egive it a try so we can merge it.

rmlearney-digicatapult, 2020-07-02:
Thank you @facchinm ! I'll try tonight.

closed 2020-07-03 by facchinm