arduino-libraries/ArduinoBLE #66: disconnect() does not trigger callback

rmlearney-digicatapult, 2020-04-07:
Example:

1. Initialise BLE
2. Create callbacks for central connection and disconnection to turn LED on and off
3. Connect to central
4. disconnect() after 10 seconds
5. Readvertise and allow reconnection

Expected behaviour - disconnection from central should trigger callback and turn off LED
Actual behaviour - disconnection from central with disconnect() method does not trigger callback

```c++
#include <ArduinoBLE.h>

BLEService TestService("DEAD");
BLECharacteristic TestData("BEEF", BLEIndicate | BLENotify, 2, true);
BLEDescriptor TestDescriptor("BEEF", "Test");

uint32_t genericTimer = 0;
bool wasConnected = false;

void setup(){

  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  initBLE();
}

void loop(){
  BLEDevice central = BLE.central(); // begin listening for centrals to connect

  if(central){
    genericTimer = millis();
    while(central.connected()){
      if(millis() - genericTimer >= 10000){ // Wait 10 seconds
        BLE.disconnect(); // Disconnect BLE - LED should go out
        wasConnected = true;
      }
    }
  }

  if(wasConnected){
    genericTimer = millis();
    while(millis() - genericTimer <= 15000){} // Wait 15 seconds
    wasConnected = false;
    BLE.advertise();
  }
}

void initBLE(void){ 
  BLE.begin();
  BLE.setEventHandler(BLEConnected, blePeripheralConnectHandler);
  BLE.setEventHandler(BLEDisconnected, blePeripheralDisconnectHandler);
  BLE.setLocalName("TestName");
  BLE.setDeviceName("Test Device Name");
  TestService.addCharacteristic(TestData);
  TestData.addDescriptor(TestDescriptor);
  BLE.addService(TestService);
  BLE.setAdvertisedService(TestService);
  BLE.advertise();
}

void blePeripheralConnectHandler(BLEDevice central){
    // turn on the LED to indicate the connection:
    digitalWrite(LED_BUILTIN, HIGH);
}

void blePeripheralDisconnectHandler(BLEDevice central){
    // when the central disconnects, turn off the LED:
    digitalWrite(LED_BUILTIN, LOW);
}
```

rmlearney-digicatapult, 2020-04-07:
Looks like the `BLEDisconnect` callback is triggered from ATTClass::removeConnection() but not ATTClass::disconnect()

Polldo, 2020-07-08:
Hi @rmlearney-digicatapult ,
actually the callback is expected to be triggered only by disconnection events coming from the HCI (i.e. caused by other devices).
If it was triggered even upon user side disconnection, the two events would not be distinguishable.
So, if You want to react to a disconnection under your control, You can directly check the return value of the disconnect call with something like:
```C++
if (BLE.disconnect()) {
    digitalWrite(LED_BUILTIN, LOW);
}
```
Similarly, the same approach is used for connection callbacks. If the connection is coming from other devices the callback is triggered, while if the connection is requested by our device the callback is not triggered.

rmlearney-digicatapult, 2020-07-08:
Hi @Polldo may I suggest you deprecate `blePeripheralDisconnectHandler` in that case?

And why would the user care about the source of disconnection when wanting to trigger a callback? 

Could you maybe give me a use case, because I just want to use `blePeripheralDisconnectHandler` to handle any disconnect events.

Polldo, 2020-07-09:
Hi @rmlearney-digicatapult ,
distinguishing between disconnection sources could be very useful. For example, a central could react to a disconnection due to a too weak signal by alerting the user in some way or by trying to reconnect.
However, I see your point. One way to have both features could be to propagate a disconnection reason.
Anyway, thanks for pointing this out. I'll think about it.

rmlearney-digicatapult, 2020-07-09:
Thank you @Polldo - I completely agree that the reason for disconnection would be useful information