arduino-libraries/ArduinoBLE #68: Found two problems on ArduinoBLE.h with Arduino Nano 33 BLE Sense

qazqwe345, 2020-04-11:
1. Use BLE.setConnectionInterval(40, 40), the theoretical communication rate is 1/(40ms*1.25) = 20 Hz, but I test this funtion physically, its communication rate is actually only 10 Hz.
setConnectionInterval()
Description
Set the minimum and maximum desired connection intervals in units of 1.25 ms.

2. Use BLE.setConnectionInterval(8, 8), but the communication rate is limited to 40 Hz maximumly. but the theoretical communication rate should be 100Hz and the BLE module on Arduino Nano 33 BLE Sense should support up to 800Hz maximumly, even though I set BLE.setConnectionInterval(4, 4) or less connection interval, the communication rate is limited to 40 Hz (already test on Arduino Nano 33 BLE Sense to Arduino Nano 33 BLE Sense, Arduino Nano 33 BLE Sense to win10 PC, Arduino Nano 33 BLE Sense to Raspberry Pi 3 B+, all of them is limited to the communication rate of 40 Hz))

rmlearney-digicatapult, 2020-04-12:
Do you have a sample sketch so we can see and test?

qazqwe345, 2020-04-14:
[central_and_peripheral.zip](https://github.com/arduino-libraries/ArduinoBLE/files/4474207/central_and_peripheral.zip)

qazqwe345, 2020-04-14:
peripheral send data and central collect data.
open the serial monitor of central will see the delta time of two send

Naphier, 2020-07-16:
If you upload your code as a code snippet and no a zip then you're more likely to get some assistance.

Polldo, 2020-07-24:
Hi @qazqwe345 ,
I've made a test using a nano33iot as a peripheral and a smartphone as central using the nordic nRF Connect application. Then, I also tried your sketches using two nano33iot.
As far as I can see, the connection interval is correctly negotiated between the central and the peripheral, but performing a read request, as you did in the sketch, seems to take more than 2 connection events. So there is probably some overhead in preparing the read response.  
Also, from ble specs, the allowed range for the connection interval is 7.5 ms to 4 s. Thus, you cannot use a value smaller than 6 for the min connection interval.