arduino-libraries/ArduinoBLE #70: Transferring 512 byte packets of data

mattleesmi, 2020-04-20:
Hello,

A colleague and I have been working on a research project that uses the Nano 33 BLE Senses and IoTs. As part of this project, we need to transfer data logs between SD cards (BLE Sense to IoT) over Bluetooth.  We have managed to get transfer working well however we noticed that packets over 246 bytes (approximately) become "garbled" beyond/around the 246th byte.

What is curious is that if the message is received by a smartphone (from the Sense) the full 512 bytes are received correctly, whereas the IoT receives a garbled message. I have attached an example of the message error. 
[Garbled_Message.txt](https://github.com/arduino-libraries/ArduinoBLE/files/4504646/Garbled_Message.txt)

My colleague traced the issue to the function:
void HCIClass::handleAclDataPkt(uint8_t /*plen*/, uint8_t pdata[])

However, he informed me that he could not go any deeper as he can't access the firmware. When he tested it he used a raspberry pi to transmit a 512 packet which was unsuccessfully collected by the IoT.

What might be the cause of this? Please let me know if you require more information, if needed I will try to portion out the code we are permitted to upload.

Thank you for your time.
Matt

gsp8181, 2020-04-23:
The firmware is located here https://github.com/arduino/nina-fw

facchinm, 2020-07-02:
May be related to this (https://github.com/arduino-libraries/ArduinoBLE/issues/73) and work by a chance in 33BLE ?
@mattleesmi it might be worth to try the patch in the issue I linked and report back, thanks.

Polldo, 2020-10-26:
Hi @mattleesmi .
Here the problem is that the BLE controller of the nano33ble has a MTU of 247. So you cannot receive/transmit more than that in a single packet. In order to receive all the 512 bytes, you should split them in more packets.

JAICHANGPARK, 2021-05-26:
Hi All 
How to set mtu this arduino ble library?

Polldo, 2021-05-26:
Hi @JAICHANGPARK ,
there is currently no API to set the MTU. You may create a new issue to request this enhancement.
However, you can try to set the MTU of the connection from another device (if the other device can perform the request: for example you can try this by using the nordic nrf app for smartphones)

closed 2021-12-16 by mattleesmi

gb88, 2023-01-26:
I create the api to set the MTU, https://github.com/gb88/ArduinoBLE