arduino-libraries/ArduinoBLE #71: subscribe()/valueUpdated() with Nano33 BLE and Adafruit BlueFruit LE Friend not working

willtmakinen78, 2020-05-01:
I've been using an Arduino Nano 33 BLE to communicate with my PC over the BlueFruit LE Friend's UART characteristic. Using the default ArduinoBLE library, I'm able to connect to the BlueFruit from the Nano33 as a central device, and write to the BlueFruit and have it show up in a serial terminal. I can get the Nano33 to read from the Bluefruit as well, by sending a Characteristic.read() request and a subsequent Charcterisitc.readValue() command. Subscribing to the rx characterisitic on the Nano33 is an issue, however, even though the Bluefruit's [info page](https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/uart-service) says that the rx characterisitc (address 0x0003) is capable of Notify. At first, Charcteristic.subscribe() would always return false, until I followed @gdsport's advice [here](https://github.com/arduino-libraries/ArduinoBLE/issues/20) and modified BLERemoteDescriptor.cpp. Now, although subscribe() returns true, Charcteristic.valueUpdated() always returns false, even when opening up a serial port to the BlueFruit and sending data with the UART service. This is the relevant Arduino  code: 

    BLECharacteristic rxCharacteristic = peripheral.characteristic("6e400003-b5a3-f393-e0a9-e50e24dcca9e");

      // make sure BlueFruit UART characteristics have been loaded properly are of
      // proper format
      if (!rxCharacteristic) {
        Serial.println("Peripheral does not have RX characteristic!");
        peripheral.disconnect();
        return;
      } else if (!rxCharacteristic.canSubscribe()) {
        Serial.println("rxCharacteristic is not subscribable!");
        peripheral.disconnect();
        return;
      } else if (!rxCharacteristic.subscribe()) {
        Serial.println("rxCharacteristic subscription failed!");
        peripheral.disconnect();
        return;
      }
      Serial.println("subscribed");

      while (peripheral.connected()) {

          if (rxCharacteristic.valueUpdated()) {
            Serial.println("value updated");
          }
    }

For reference, this is the modified code from BLERemoteDescriptor.cpp:

    int BLERemoteDescriptor::writeValue(const uint8_t value[], int length)
    {
      if (!ATT.connected(_connectionHandle)) {
        return false;
      }

      uint16_t maxLength = ATT.mtu(_connectionHandle) - 3;

      if (length > (int)maxLength) {
        // cap to MTU max length
        length = maxLength;
      }

      _value = (uint8_t*)realloc(_value, length);
      if (_value == NULL) {
        // realloc failed
        return 0;
      }  

    #if 0
      uint8_t resp[4];
      int respLength = ATT.writeReq(_connectionHandle, _handle, value, length, resp);

      if (!respLength) {
        return 0;
      }

      if (resp[0] == 0x01) {
        // error
        return 0;
      }
    #else 
	    ATT.writeCmd(_connectionHandle, _handle, value, length);
    #endif

      memcpy(_value, value, length);
      _valueLength = length;

      return 1;
    }

Any thoughts on what might be causing this?

Arduino IDE version 1.8.12 and ArduinoBLE version 1.1.2

Gerriko, 2020-05-03:
I had noticed the question raised about the failure to subscribe on the Arduino Forum and began to investigate.

For what it's worth, I discovered that within this function BLERemoteCharacteristic::writeCccd(uint16_t value) it determines the number of descriptors with this call int numDescriptors = descriptorCount();

For some reason the Bluefruit device is returning 0.

I ran the identical peripheral example using a BBC microbit and this correctly returns the value 2 (as in, there are two descriptors present for that characteristic 2901 and 2902).

willtmakinen78, 2020-05-04:
> I ran the identical peripheral example using a BBC microbit and this correctly returns the value 2 (as in, there are two descriptors present for that characteristic 2901 and 2902).
 
So would you say that it's a Nano33 BLE issue? Also I'm assuming you were able to subscribe on the BBC microbit.

Gerriko, 2020-05-04:
Yes, the Nano33 can subscribe to the BBC microbit as it can loop through the number of descriptors available and find 0x2902 to then apply the write value function. With Bluefruit, because number of descriptors is zero it bypasses and thus never updates 0x2902.

willtmakinen78, 2020-05-04:
I see - any suggestions on how to fix the descriptorCount() function? Otherwise I'm going to order an ESP32 and be done with this :/.

Gerriko, 2020-05-05:
I've discovered that this is not as straightforward as one might think. 

I created PSoC4 peripheral with the same GATT service + characteristics as the Bluefruit device but this time I did not implement the NOTIFY update functionality, just to see what happens (i.e. this should throw up an error).

The ESP32 returns that all is OK, when it should not have done so - so in this case the ESP32 would have crashed or not received any values from NOTIFY. 

The nano BLE33 board (with ArduinoBLE library) interestingly finds the 2 descriptors (2901 and 2902) and says that it was able to update 2902 but then does not update the "fallback" option as the "if (_properties & (BLENotify | BLEIndicate))" was not true  (with the Bluefruit device this was true but the cccd.writeValue must not return true for what ever reason).

`  if (_properties & (BLENotify | BLEIndicate)) {
    // no CCCD descriptor found, fallback to _valueHandle + 1
    BLERemoteDescriptor cccd(NULL, 0, _connectionHandle, _valueHandle + 1);
   return cccd.writeValue((uint8_t*)&value, sizeof(value));
  }`

Gerriko, 2020-05-05:
Debug using BLE.debug(Serial) - nano BLE connecting with Bluefruit:

Found f7:0f:7f:2d:4e:56 'Adafruit Bluefruit LE' 6e400001-b5a3-f393-e0a9-e50e24dcca9e
HCI COMMAND TX -> 010C20020000
HCI EVENT RX <- 040E04010C2000
Connecting ...
HCI COMMAND TX -> 010D2019600030000001564E2D7F0FF70006000C000000C80004000600
HCI EVENT RX <- 040F0400010D20
HCI EVENT RX <- 043E13010000000001564E2D7F0FF708000000C80007
Connected
Discovering attributes ...
HCI ACLDATA TX -> 02000007000300040002F700
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 020020070003000400031700
HCI ACLDATA TX -> 0200000B0007000400100100FFFF0028
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002012000E000400110601000700001808000B000118
HCI ACLDATA TX -> 0200000B0007000400100C00FFFF0028
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201A001600040011140C00130023D1BCEA5F782315DEEF121230150000
HCI ACLDATA TX -> 0200000B0007000400101400FFFF0028
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200C0008000400110614001E000A18
HCI ACLDATA TX -> 0200000B0007000400101F00FFFF0028
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201A001600040011141F00FFFF9ECADC240EE5A9E093F3A3B50100406E
HCI ACLDATA TX -> 0200000B000700040008010007000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B0017000400090702000A0300002A0400020500012A0600020700042A
HCI ACLDATA TX -> 0200000B000700040008080007000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200900050004000108080001
HCI ACLDATA TX -> 0200000B00070004000808000B000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200D000900040009070900200A00052A
HCI ACLDATA TX -> 0200000B0007000400080B000B000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002009000500040001080B000A
HCI ACLDATA TX -> 0200000B0007000400080C0013000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B001700040009150D00040E0023D1BCEA5F782315DEEF121232150000
HCI ACLDATA TX -> 0200000B0007000400080F0013000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B001700040009150F0018100023D1BCEA5F782315DEEF121231150000
HCI ACLDATA TX -> 0200000B000700040008110013000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B00170004000915120002130023D1BCEA5F782315DEEF121234150000
HCI ACLDATA TX -> 0200000B000700040008140013000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200900050004000108140001
HCI ACLDATA TX -> 0200000B00070004000814001E000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B001700040009071500021600292A1700021800242A1900021A00282A
HCI ACLDATA TX -> 0200000B0007000400081B001E000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002014001000040009071B00021C00262A1D00021E00272A
HCI ACLDATA TX -> 0200000B0007000400081F001E000328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002009000500040001081F0001
HCI ACLDATA TX -> 0200000B0007000400081F00FFFF0328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B0017000400091520001021009ECADC240EE5A9E093F3A3B50300406E
HCI ACLDATA TX -> 0200000B0007000400082200FFFF0328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200201B0017000400091524000C25009ECADC240EE5A9E093F3A3B50200406E
HCI ACLDATA TX -> 0200000B0007000400082600FFFF0328
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 020020090005000400010826000A
HCI ACLDATA TX -> 020000090005000400040B000B00
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200A000600040005010B000229
HCI ACLDATA TX -> 020000090005000400040C000B00
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 02002009000500040001040C0001
HCI ACLDATA TX -> 020000090005000400042600FFFF
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200A0006000400050126000129
HCI ACLDATA TX -> 020000090005000400042700FFFF
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 020020090005000400010427000A
Attributes discovered
--- Number descriptors found: 0
HCI ACLDATA TX -> 0200000900050004001222000100
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200900050004000112220003
--- writevalue.cccd returned: 0
Fail to subscribe...
HCI ACLDATA RX <- 02002010000C000500120208001000200000002C01
HCI ACLDATA TX -> 0200000A0006000500130202000000
HCI COMMAND TX -> 0113200E00001000200000002C0104000600
HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 040F0400011320
HCI EVENT RX <- 043E0A03000000100000002C01

Gerriko, 2020-05-05:
I've found that within the writeValue function (BLERemoteDescriptor), the Bluefruit has resp[0] = 0x01, which triggers an error condition. Not sure what this means though.

willtmakinen78, 2020-05-06:
Alright, so it seems that my issues are a combination of BlueFruit and ArduinoBLE incompatabilities. Debugging the library is unfortunately a bit past the scope of project, since it's due in a week. In general I would be interested in helping out however, so please let me know if there's anything you need me to do.

Polldo, 2020-07-21:
Hi @willtmakinen78 and @Gerriko ,
thanks a lot for detailing the issue. In particular to @Gerriko for trying many combinations and posting logs!  
I should have found the problem with the library, you can check this out here https://github.com/arduino-libraries/ArduinoBLE/pull/99

Could you please try if this fix solves the issue?
Thanks!

heedongrichrich, 2022-07-27:
I am having the same issue with Nano RP4020.
Is there someone who's facing the same issue?

s-light, 2026-02-07:
just think i stumbled over this error trying to use an UNO R4 Wifi with DroidPad
https://github.com/UmerCodez/DroidPad/issues/47

also posted this in the arduino forum
https://forum.arduino.cc/t/arduinoble-subscribe-fails/1429630

the code in the library in the meantime changed a lot.
and as i have no idea about the deep-internals of Bluetooth i have no idea where to start debugging..

here is my debug output from the 
https://github.com/Make-Your-School/mks-Arduino-UNO_R4_WiFi/blob/main/examples/dev/UNO_R4_WiFi-Bluetooth_DroidPad/UNO_R4_WiFi-Bluetooth_DroidPad.ino
sketch:
```
UNO_R4_WiFi-Bluetooth_DroidPad.ino
HCI COMMAND TX -> 01030C00
HCI EVENT RX <- 040E0405030C00
HCI COMMAND TX -> 01011000
HCI EVENT RX <- 040E0C050110000916000960001600
HCI COMMAND TX -> 01010C08FFFFFFFFFFFFFF3F
HCI EVENT RX <- 040E0405010C00
HCI COMMAND TX -> 01012008FF03000000000000
HCI EVENT RX <- 040E0405012000
HCI COMMAND TX -> 01022000
HCI EVENT RX <- 040E0705022000FB000C
HCI COMMAND TX -> 010C20020001
HCI EVENT RX <- 040E04050C2000
HCI COMMAND TX -> 010B200701200020000000
HCI EVENT RX <- 040E04050B2000
HCI COMMAND TX -> 010C20020101
HCI EVENT RX <- 040E04050C2000
Bluethooth Device namens 'HackBLE' fertig initialisiert.
HCI EVENT RX <- 043E2B02010301B64197B168111F1E16F3FE4A17235941353711326E44026457387658497C1DBA2E2DD2BB5894CF
HCI EVENT RX <- 043E1F02010001399CE180C65B130201020F094661697270686F6E652034203547C4
HCI EVENT RX <- 043E0C02010401399CE180C65B00C4
HCI EVENT RX <- 043E1D02010000BE3133445510110201050302E0FF09FFFFFF1055443331BEBC
HCI EVENT RX <- 043E2002010400BE31334455101413094E574A323032313037303930313030313811BD
HCI EVENT RX <- 043E1D02010000272233445510110201050302E0FF09FFFFFF105544332227C2
HCI EVENT RX <- 043E20020104002722334455101413094E574A323032313037303930313030333411C3
HCI EVENT RX <- 043E2602010001B16C6464BD4C1A0201020409467034110711B1A440EA62E1AFAB4409F5D7C1BF4FCC
HCI EVENT RX <- 043E0C02010401B16C6464BD4C00CC
Gefunden: 'Fp4' Service UUID '4fbfc1d7-f509-44ab-afe1-62ea40a4b111'
HCI COMMAND TX -> 010C20020000
HCI EVENT RX <- 040E04050C2000
Verbinden ...
HCI COMMAND TX -> 010D2019600030000001B16C6464BD4C0006000C000000C80004000600
HCI EVENT RX <- 040F0400050D20
HCI EVENT RX <- 043E1F0A0001000001B16C6464BD4C0000000000000000000000000C000000C80000
Verbunden
suche Attribute ...
HCI ACLDATA TX -> 02010007000300040002F200
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 0201200B00070004001D03000100FFFF
HCI ACLDATA TX -> 0201000500010004001E
HCI ACLDATA RX <- 020120070003000400030502
HCI ACLDATA RX <- 0201200600020006000B2D
HCI EVENT RX <- 043E0B0701001B004801F6004808
HCI EVENT RX <- 0413050101000100
HCI ACLDATA TX -> 0201000B0007000400100100FFFF0028
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 02012012000E000400110601000900011814001D000018
HCI ACLDATA TX -> 0201000B0007000400101E00FFFF0028
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 02012092008E00040011148600950011B1A440EA62E1AFAB4409F5D7C1BF4F9600A60011B1A440EA62E1AFAB4409F5D7C1BF4FB900BD0011B1A440EA62E1AFAB4409F5D7C1BF4FBE00C10011B1A440EA62E1AFAB4409F5D7C1BF4FC200C60011B1A440EA62E1AFAB4409F5D7C1BF4FC700CC0011B1A440EA62E1AFAB4409F5D7C1BF4FCD00FFFF11B1A440EA62E1AFAB4409F5D7C1BF4F
HCI ACLDATA TX -> 0201000B000700040008010009000328
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 02012022001E00040009070200200300052A04000205003A2B06000A0700292B08000209002A2B
HCI ACLDATA TX -> 0201000B0007000400080A0009000328
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 02012009000500040001080A0001
HCI ACLDATA TX -> 0201000B00070004000814001D000328
[... 233 lines cut ...]
HCI ACLDATA TX -> 02010009000500040004D100FFFF
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 0201200A00060004000501D1000229
HCI ACLDATA TX -> 02010009000500040004D200FFFF
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 0201200A00060004000501D2000229
HCI ACLDATA TX -> 02010009000500040004D300FFFF
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 0201200A00060004000501D3000229
HCI ACLDATA TX -> 02010009000500040004D400FFFF
HCI EVENT RX <- 0413050101000100
HCI ACLDATA RX <- 0201200900050004000104D4000A
Attribute gefunden
DroidPad characteristic nicht lesbar!
DroidPad characteristic nicht schreibbar!
DroidPad characteristic abonnierbar..
HCI ACLDATA TX -> 0201000900050004001289000100
HCI EVENT RX <- 0413050101000100
subscription failed!
HCI COMMAND TX -> 01060403010013
HCI EVENT RX <- 040F0400050604
HCI EVENT RX <- 04050400010016
Peripheral disconnected

```