arduino-libraries/ArduinoBLE #74: Pheripheral connection lost / frozen after multiple connect / disconnects from central

AndreasLamparter, 2020-05-07:
Hello,

i have a simple app with a arduino nano ble and a android device.

The android device search the arduino nano and show the pheripheral. The android app reads a value from the nano device. After multiple connect and disconnects on the android app the arduino nano is not reachable. Only a restart solves the problem.

How can i debug the problem and can get more information about the problem?

Best regards
Andreas

AndreasLamparter, 2020-05-11:
Here is the BLE debug output. After disconnect a device, the android ble scan doesn't find the device anymore.

Init:
HCI COMMAND TX -> 01030C00 
HCI EVENT RX <- 040E0401030C00
HCI COMMAND TX -> 01011000    
HCI EVENT RX <- 040E0C01011000090323095F000323
HCI COMMAND TX -> 01010C08FFFFFFFFFFFFFF3F    
HCI EVENT RX <- 040E0401010C00
HCI COMMAND TX -> 01022000    
HCI EVENT RX <- 040E0701022000000104
HCI COMMAND TX -> 0106200FA000A0000000000000000000000700
HCI EVENT RX <- 040E0401062000
HCI COMMAND TX -> 010820200702010603021818000000000000000000000000000000000000000000000000
HCI EVENT RX <- 040E0401082000
HCI COMMAND TX -> 010920200C0B09414C5F4379636C696E6700000000000000000000000000000000000000
HCI EVENT RX <- 040E0401092000
HCI COMMAND TX -> 010A200101
HCI EVENT RX <- 040E04010A2000

Connect:
Device connected 4c:ef:b8:7e:c2:b0
HCI EVENT RX <- 043E13010000000101B0C27EB8EF4C27000000D00705

HCI ACLDATA RX <- 0200200900050004001210000100
HCI ACLDATA TX -> 0200000900050004000112100003

HCI ACLDATA RX <- 0200200900050004001215000100
HCI ACLDATA TX -> 0200000900050004000112150003
HCI EVENT RX <- 0413050100000100

HCI EVENT RX <- 0413050100000100

Disconnect
Device disconnected 4c:ef:b8:7e:c2:b0
Central disconnected
HCI EVENT RX <- 04050400000013
HCI COMMAND TX -> 010A200101
HCI EVENT RX <- 040E04010A2000

mattleesmi, 2020-05-11:
Hello AndreasLamparter,

Does the board freeze/crash?

AndreasLamparter, 2020-05-11:
Hello,

the device does not crash, i can print continuous messages in the loop() function.

mattleesmi, 2020-05-11:
Hello AndreasLamparter,

Are you using a Nano BLE/BLE Sense instead of a Nano IoT? People have had some disconnection issues with the BLE, (for example here: https://forum.arduino.cc/index.php?topic=645907.0). I know that isn't much help but you are not alone!

AndreasLamparter, 2020-05-11:
Hello mattleesmi,

i bought following Arduino Nano 33
BLE https://www.reichelt.de/arduino-nano-33-ble-nrf52840-ohne-header-ard-nano-33ble-p261304.html?PROVID=2788&gclid=EAIaIQobChMInaGhlI2s6QIViud3Ch2XRgxjEAQYAiABEgKBlvD_BwE&&r=1

Best regards 
Andreas

AndreasLamparter, 2020-05-11:
Hello mattleesmi,

i work with VSCode and paltform.io

Environment:
platform = nordicnrf52
board = nano33ble
framework = arduino

Libraries:
extEEPROM
ArduinoBLE
MPU6050_light
ArduinoLog

Best regards
Andreas

AndreasLamparter, 2020-05-11:
Close the issue

please see:
https://forum.arduino.cc/index.php?topic=645907.0
https://github.com/arduino-libraries/ArduinoBLE/issues/45

closed 2020-05-11 by AndreasLamparter

martinstingl, 2020-12-22:
@AndreasLamparter 
Why did you close this issue? 
According to the information given in your linked forum topic there is a workaround using a watchdog timer, but the real issue according to the user Klaus_K is:

"I have a simple test program which I used on the Arduino Nano 33 IoT and Nano 33 BLE.

The behavior for them is different. I guess they both use the same BLE library but there might be differences in the integration with mbedOS on the Nano 33 BLE.

**The Arduino Nano 33 BLE does not set the central.connected correctly** when the connection is lost. It returns central.rssi() = 0 a couple of times when the connection is about to get lost. But it can recover from that when the connection gets better. But when the connection is lost it cannot recover without reset.

The Arduino Nano 33 IoT does set central.connected correctly so the sketch can get back to a state where it waits for connections. The central.rssi never gets to 0, but gets to some maximum negative value around -128 or so as expected."