arduino-libraries/ArduinoBLE #77: Arduino Nano BLE 33 Disconnecting/Reconnecting while uploading

Dozerhank, 2020-05-18:
When I upload to my Arduino Nano, after it compiles the USB disconnects then reconnects. This message appears afterwards:

Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash

Done in 0.000 seconds
Write 76396 bytes to flash (19 pages)

[                              ] 0% (0/19 pages)
[=                             ] 5% (1/19 pages)
[===                           ] 10% (2/19 pages)
[====                          ] 15% (3/19 pages)
[======                        ] 21% (4/19 pages)
[=======                       ] 26% (5/19 pages)
[=========                     ] 31% (6/19 pages)
[===========                   ] 36% (7/19 pages)
[============                  ] 42% (8/19 pages)
[==============                ] 47% (9/19 pages)
[===============               ] 52% (10/19 pages)
[=================             ] 57% (11/19 pages)
[==================            ] 63% (12/19 pages)
[====================          ] 68% (13/19 pages)
[======================        ] 73% (14/19 pages)
[=======================       ] 78% (15/19 pages)
[=========================     ] 84% (16/19 pages)
[==========================    ] 89% (17/19 pages)
[============================  ] 94% (18/19 pages)
[==============================] 100% (19/19 pages)
Done in 2.967 seconds

The program doesn't run after this, even a simple LED blinking code doesn't.

morettigiorgio, 2020-05-18:
Did you try uploading in bootload mode? You have to press the button twice quickly.
Are you using the WDT?

facchinm, 2020-07-02:
Hi @Dozerhank ,
would you mind moving the issue under https://github.com/arduino/ArduinoCore-nRF528x-mbedos ? Thanks

closed 2020-07-02 by facchinm