arduino-libraries/ArduinoBLE #78: Arduino Nano 33 BLE Bluetooth stops working after some minutes

AlexanderFillbrunn, 2020-05-20:
Hi,
I have written the below code for an Arduino Nano 33 BLE, which is hooked up to an HX711 to read values from a load cell, and an LCD to output the current weight on the cell. Additionally, the weight is sent to a connected device via Bluetooth, if one is connected. As long as no Bluetooth device is connected, the weight is updated on the LCD correctly and continuously and everything is good. Once I connect my phone via Bluetooth, e.g. using the nRF Connect by Nordic Semiconductors, and listen for the characteristic, the characteristic updates for a couple of seconds and then the Arduino is stuck completely. It updates neither the display nor the characteristic. Am I using the library incorrectly, or is there another problem?

Arduino model: Nano 33 BLE
ArduinoBLE version: 1.1.2
LiquidCrystal version: 1.0.7
HX711 version: 0.7.4

If you need any other information or if I can debug anything, please let me know!

```
#include <ArduinoBLE.h>
#include "HX711.h"
#include <LiquidCrystal.h>

#define DOUT  3
#define CLK  2

LiquidCrystal lcd(12, 11, 7, 6, 5, 4);
HX711 scale;
float calibration_factor = 16000;
BLEService myService("F5A1287A-227D-5C9E-AD2C-11D0FD6ED640");
BLEFloatCharacteristic myChar("F5A1287A-227D-5C9E-AD2C-11D0FD6ED641", BLERead | BLENotify);

void setup() {
  Serial.begin(9600);
  Serial.println("Test Sketch");
  // set up the LCD's number of columns and rows:
  lcd.begin(16, 2);
  // Scale
  scale.begin(DOUT, CLK);
  scale.tare();
  scale.set_scale(calibration_factor);

  // Bluetooth
  if (!BLE.begin()) 
  {
    Serial.println("starting BLE failed!");
    while (true);
  }
  BLE.setLocalName("Test");
  BLE.setAdvertisedService(myService);
  myService.addCharacteristic(myChar);
  BLE.addService(myService);
  
  BLE.advertise();
  Serial.println("Bluetooth device active, waiting for connections...");
}

void loop() {
  BLEDevice central = BLE.central();
  if (central) {
      Serial.print("Connected to central: ");
      Serial.println(central.address());

      while (central.connected()) {
        float force = readAndPrintForce(true);
        myChar.writeValue(force);
        delay(500);
     }
     Serial.print("Disconnected from central: ");
     Serial.println(central.address());
  } else {
    readAndPrintForce(false);
  }
  delay(500);
}

float readAndPrintForce(bool connected)
{
  float force = scale.get_units(3);
  Serial.print(force);
  Serial.println("kg");
  
  lcd.clear();
  lcd.print(force, 2);
  lcd.setCursor(0, 1);
  if (connected) {
    lcd.print("Connected");
  } else {
    lcd.print("Not connected");
  }
  return force;
}
```

AlexanderFillbrunn, 2020-05-20:
I removed all other libraries and only kept the bluetooth code, sending random floats, and could not observe the described behavior anymore. I will therefore close this issue for now and try to find out whats's wrong.

closed 2020-05-20 by AlexanderFillbrunn

morettigiorgio, 2020-05-21:
try to comment out interrupt line into HX711 library

AlexanderFillbrunn, 2020-05-21:
> try to comment out interrupt line into HX711 library

Hi @morettigiorgio, thanks for your response! Do you mean the noInterrupt() and interrupt() lines, or just one of them? Doesn't that have nasty side effects when reading from the scale?

morettigiorgio, 2020-05-21:
Yes, both lines and yes, it is possible, but in my case it is only way for resolve the problem!
I have to say that sometimes it happens to read incorrect values, I'm not sure about the comments line made, but I solved by averaging the values of HX

AlexanderFillbrunn, 2020-05-21:
Thanks @morettigiorgio! I simply switched to the [HX711_ADC ](https://github.com/olkal/HX711_ADC)library. Here it works fine.