arduino-libraries/ArduinoBLE #82: my Eddystone UID test example has no service data

Gerriko, 2020-05-29:
I created an Eddystone UID example, but it is not working correctly. Please see my post on Arduino Forum: https://forum.arduino.cc/index.php?topic=687074.0

From what I can see is that when I use "setLocalName" this data is added to the service data.

I had a look at the library. In the GAPClass::GAPClass()  there is  _localName(NULL) and this variable is set via this function. This in turn is added to the scanResponseData.

Now, looking at the GAPClass::GAPClass() there is in fact  _serviceData(NULL) and _serviceDataLength(0). These two variables are never set, or I could find where they are set. Thus as fas as I can see, this logic fails within the function int GAPClass::advertise():

`  if (_serviceData && _serviceDataLength > 0 && advertisingDataLen >= (_serviceDataLength + 4)) {
    advertisingData[advertisingDataLen++] = _serviceDataLength + 3;
    advertisingData[advertisingDataLen++] = 0x16;

    memcpy(&advertisingData[advertisingDataLen], &_serviceDataUuid, sizeof(_serviceDataUuid));
    advertisingDataLen += sizeof(_serviceDataUuid);

    memcpy(&advertisingData[advertisingDataLen],_serviceData, _serviceDataLength);
    advertisingDataLen += _serviceDataLength;
  }`

Is this intentional?

My understanding is that the Eddystone format is serviceData and not scanResponseData. As such it cannot work as it stands.

Gerriko, 2020-06-01:
I am looking at this line of code again (int GAPClass::advertise()) in GAP.cpp:

_if (_serviceData != NULL && _serviceDataLength > 0 && advertisingDataLen >= (_serviceDataLength + 4)) { ...}

I think this should read...

_if (_serviceData != NULL && _serviceDataLength > 0 && **31 >= (_serviceDataLength + 4)**) { ...}

As in you, do never want to have (serviceDataLength+4) to be greater than the allowable max advertising length which is 31 bytes....

If you had used a const for max Advertising length then this would have been easier to spot... I see you hard code the value "31" in quite a few places.

Gerriko, 2020-06-02:
I too made an error. After rechecking again I believe that line of code should read:

_if (_serviceData != NULL && _serviceDataLength > 0 &&  (advertisingDataLen 
 + _serviceDataLength + 4) <= 31) { ... }

rmlearney-digicatapult, 2020-06-29:
If this fixes your issue, would you submit a PR for others who might come across this problem in the future?

Polldo, 2020-08-17:
Fixed. See https://github.com/arduino-libraries/ArduinoBLE/commit/633e1aa049549fa5b8f60bd9e0baae6f8f5c4d37

closed 2020-08-17 by Polldo