arduino-libraries/ArduinoBLE #84: BLE no longer compiles

f0rodo, 2020-06-03:
I am no longer able to compile the library 

```
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp: In function 'void bleLoop()':
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp:112:11: error: 'LowPowerTimer' is not a member of 'mbed'
     mbed::LowPowerTimer timer;
           ^~~~~~~~~~~~~
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp:115:38: error: 'timer' was not declared in this scope
         last_update_us += (uint64_t) timer.read_high_resolution_us();
                                      ^~~~~
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp:115:38: note: suggested alternative: 'time'
         last_update_us += (uint64_t) timer.read_high_resolution_us();
                                      ^~~~~
                                      time
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp:131:19: error: 'CriticalSectionLock' is not a member of 'mbed'
             mbed::CriticalSectionLock critical_section;
                   ^~~~~~~~~~~~~~~~~~~
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp:152:15: error: 'wait_us' was not declared in this scope
               wait_us(wait_time_us);
               ^~~~~~~
Arduino/libraries/ArduinoBLE/src/utility/HCICordioTransport.cpp:152:15: note: suggested alternative: 'init_wsf'
               wait_us(wait_time_us);
               ^~~~~~~
               init_wsf

```

I am not sure if something changed.

facchinm, 2020-06-03:
Hi @f0rodo ,
which version of mbed core are you running?

f0rodo, 2020-06-03:
I am not sure. How do I determine that? I use the arduino ide and arduino-cli

f0rodo, 2020-06-03:
I have a partial fixe I added the mbed.h to the src/utility/HCICordioTransport.cpp and it seems to work.

facchinm, 2020-06-03:
Yes, I fixed it here https://github.com/arduino-libraries/ArduinoBLE/commit/8a49b8c1e0ac195db38a0639378a1b50b7bdd566#diff-dd7fc9ada03075ba058701943bb38d00R25 but it's only for mbed cores with Portenta H7 support (not yet superofficial). We'll release a new version soon, in the meantime I'm closing the issue

closed 2020-06-03 by facchinm

f0rodo, 2020-06-03:
@facchinm thank you! I was actually doing this on the nano 33 ble sense. I also have the portenta so I want to make it work on that too! Thanks! How do I use the Portenta H7 wouldn't mind testing it out.