arduino-libraries/ArduinoBLE #88: Weak Signal Disconnect not returning to main loop() - Nano BLE Sense

m-stander, 2020-06-21:
Using the Nano BLE Sense as a peripheral: 

### Issue:
A disconnect caused by a weak signal, like walking away from the Nano,  triggers the Disconnect Handler and the main loop() stops running.  

The event handlers continue to function.  Trying to reconnect will run the Connect Handler, but nothing happens after the connection because loop() is no longer running.  

(Possibly relevant: when this disconnect occurs, the RSSI usually reads 0.)

### Experimenting with other disconnect scenarios:
A disconnect caused by closing the iOS app, triggers the Disconnect Handler but the main loop() continues to run as expected.

When back in range, the iOS app can reconnect to the Nano and program functions correctly.   

(Possibly relevant: when this disconnect occurs, the RSSI reads 127.)

**Work Around:**  In the main loop, I check RSSI and if it gets below -80, I call BLE.disconnect(). 

This will invoke the Disconnect Handler and the main program loop will continue to run.  

When back in range, the iOS app can reconnect to the Nano and everything functions properly.  

This Work Around keeps the Nano functioning, but does reduce the expected BLE range. 

### Additional Details:
I was having problems with this issue running ArduinoBLE 1.1.2, so I downloaded the latest .zip from GitHub today (6/21/2020).  Most of my testing and the notes above refer to the GitHub version.

rmlearney-digicatapult, 2020-06-29:
There's currently no callback handling for disconnect if you're relying on that functionality. Please see issue #66 and try my PR #67. If it works for you please mention it here.

There's also a fault where you can't restart the BLE subsystem after calling .end - see issue #65. 

The Arduino team haven't commented on that issue yet, which is quite a serious bug.

facchinm, 2020-07-02:
Condensing all the issues related to weak signal in https://github.com/arduino-libraries/ArduinoBLE/issues/45 , closing therefore this one.

closed 2020-07-02 by facchinm