arduino-libraries/ArduinoBLE #91: Unsupported board?

GrizzlyMiki, 2020-07-01:
Hello! 
First of all, sorry if this is not the right way to do this, this is the first issue I post.

I was trying to use AndroidBLE with my new board and I'm getting this error:

`WARNING: library ArduinoBLE-master claims to run on samd, megaavr, mbed architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
/home/user/Arduino/libraries/ArduinoBLE-master/src/utility/HCIUartTransport.cpp:33:2: error: #error "Unsupported board selected!"
 #error "Unsupported board selected!"
  ^~~~~
/home/user/Arduino/libraries/ArduinoBLE-master/src/utility/HCIUartTransport.cpp:99:40: error: 'SerialHCI' was not declared in this scope
 HCIUartTransportClass HCIUartTransport(SerialHCI, 912600);
                                        ^~~~~~~~~
/home/user/Arduino/libraries/ArduinoBLE-master/src/utility/HCIUartTransport.cpp:99:40: note: suggested alternative: 'Serial'
 HCIUartTransportClass HCIUartTransport(SerialHCI, 912600);
                                        ^~~~~~~~~
                                        Serial
exit status 1
Error compiling for board Arduino Nano.`

If anyone knew where the problem is it would be really nice, It might be something really stupid but right now I cannot figure it out. I hope I gave enough info but if you need more just tell me and I will try my best to get it.

I leave here my specs for if they are useful:
The board is from Keywish, and it's compatible with Arduino Nano BLE V3.0 (or so the amazon page says, if you want I can post the link). 
As the board documentation for the board says, I select "Arduino Nano" as the board in the IDE.
IDE version 1.8.13.
I downloaded the ArduinoBLE library again today so I should have the 1.14 version, although Arduino's lib manager says I have the version 1.13.

facchinm, 2020-07-02:
Hi @GrizzlyMiki ,
the board you purchased is compatible with old Nano board (v3), not the [BLE](https://store.arduino.cc/arduino-nano-33-ble) one, which features a totally different processor in the same form factor.

closed 2020-07-02 by facchinm

GrizzlyMiki, 2020-07-02:
Oh... Well, thank you very much for taking the time to answer and being this fast. And sorry for creating an issue for such a silly problem.