arduino-libraries/ArduinoBLE #93: Cannot discover bluetooth device

aidanthewiz, 2020-07-04:
I have an [8Bitdo Controller](https://www.8bitdo.com/sn30-pro-plus/) that is not being discovered by the ArduinoBLE library.
The device uses Bluetooth®4.0 which is stated as supported with the ArduinoBLE library on the [product page](https://store.arduino.cc/usa/arduino-uno-wifi-rev2). 

I am using the latest NINA firmware (1.3.0) and the latest ArduinoBLE library (1.1.3). All my other Bluetooth devices show up using the PeriphrealExplorer example, except the controller which I intended to use the board with. Any help would be great!

aidanthewiz, 2020-07-05:
I found an issue in the nina-fw repository (arduino/nina-fw#35) that somewhat matches my issues. Bluetooth®4.0 HID is what I am looking for, but the ability to connect to a Bluetooth®4.0 device, in general, is close enough as I can _hopefully_ figure out HID myself.

If the issue presented in nina-fw is true, why does the [product page for the Arduino Uno WiFi Rev 2](https://store.arduino.cc/usa/arduino-uno-wifi-rev2) and the [library reference page for ArduinoBLE](https://www.arduino.cc/en/Reference/ArduinoBLE) currently claim Bluetooth®4.0 support? Or is it supported by the hardware, but not the firmware currently? Or is it only certain devices? Or maybe this is actually an issue with the library.

Overall, I might be misunderstanding or missing something, but the information is slightly scattered and the nina-fw issue#35 is conflicting, any help would be appreciated!

facchinm, 2020-07-06:
Hi @aidanthewiz ,
ArduinoBLE library only supports the low energy (hence BLE) part of the Bluetooth 4.0 protocol.
While the hardware on the UNO WIFi rev2 supports also Bluetooth Classic we don't support it directly. Using https://github.com/espressif/esp-idf would allow you to communicate with the Bluetooth peripheral; you can check this repo https://github.com/joonazan/nina-fast-bluetooth as a starting point.
Hope it helps!

closed 2020-07-06 by facchinm