arduino-libraries/ArduinoBLE #97: [MKR 1010] - Pair with

Davidffry, 2020-07-16:
Hi, 

I would like to pair my smartphone and my arduino mkr 1010, it seems that it couldn't to be pairing, can you confirm it?

Naphier, 2020-07-16:
BLE (Bluetooth Low Energy) doesn't require "pairing". It's done via device discovery -> service discovery -> characteristic discovery & read/listen or write. The "LightBlue" app on the play store is really helpful in testing. 

Pairing doesn't work with the BLE library. If you want pairing then you'll probably want to use Bluetooth classic via the Serial port class. Most BLE chips for arduino don't ship with Bluetooth Classic enabled. There may be a way to change that via firmware, but I doubt that's what you want.

If you've already attempted to pair the device with your phone, remove it from the paired device lists. Explore its BLE services and characteristics via LightBlue app.

Davidffry, 2020-07-17:
Thanks for your answer, you're right, no pairing works with this library.

to use the ArduinoBLE with smartphone, I create an android app with MIT App Inventor, and it works : 
[image]

now I search how to write to mkr or read from mkr, any ideas?

closed 2020-07-17 by Davidffry