arduino-libraries/ArduinoBLE #104: Device Name is not changing

WB3Tech, 2020-08-11:
For example, I set my device name as "A", then change it to "B", the device name stays "A".

I have validated that the local name does not have this same issue.  I have some devices that, after some time, the Device name will change from "A" to "B".  For example, after some time (15 - 30 mins) the name will finally change from "A" to "B".

Polldo, 2020-08-11:
Hi @WB3Tech ,
I've tried to replicate your issue with the minimal sketch posted below. However, I did not encounter problems: the device name is changed correctly after every 5 seconds.
Setup used: nano 33 iot with the following sketch as a peripheral. A smartphone, with the nordic nRF connect app, connected as a central to read the nano33iot device name.
Can you please try this sketch reporting here the results you obtain and detailing your setup?
Thanks
``` C++
#include "ArduinoBLE.h"

const char *names[] = {"A", "B", "C"};

void setup()
{
  BLE.begin();
  BLE.setDeviceName("A");
  BLE.setLocalName("Device-Name-Test");
  BLE.advertise();
}

void loop()
{
  static auto timeRef = millis();
  static auto count = 0;

  BLE.poll();
  if (millis() - timeRef > 5000) {
    count++;
    timeRef = millis();
    auto idx = count % 3;
    BLE.setDeviceName(names[idx]);
  }
}
```

WB3Tech, 2020-08-11:
@Polldo, yes, I will try this shortly.

MCXIV, 2021-06-09:
> Hi @WB3Tech ,
> I've tried to replicate your issue with the minimal sketch posted below. However, I did not encounter problems: the device name is changed correctly after every 5 seconds.
> Setup used: [...]

Hi, sorry for updating this, but I confirm that this code works.

But is there any possibilities to update the local name with `BLE.setLocalName()` dynamically just the way you showed it with the device name? I can't find a way to do it other than reseting the Arduino.

Thank you!

Quentin

Polldo, 2021-06-09:
Hi @MCXIV , 
yes, you can. 
Each time you change some advertising parameters, like the local name, you then need to call the advertise function `BLE.advertise()` again.
So, in your case, after updating the local name through `BLE.setLocalName()` you need to call `BLE.advertise()`

MCXIV, 2021-06-09:
Hi @Polldo,

Unfortunately, it seems that it doesn't work.

Let me explain in my example : 

My Arduino (MKR WiFi 1010) is set with X characteristics with Y services. During the setup of the BLE, I use both of these command : 

```
BLE.setLocalName(bleName);
BLE.setDeviceName(bleName);
```

bleName, in our example, is : 

`const char bleName[20] = xxx;` with each x a value read from a memory (M24C64, over I²C)

With the first init of the BLE, both of these names works perfectly. But after X seconds, using `ArduinoLowPower`, the Arduino goes into sleep mode. When waking up the Arduino, the `bleName` is read from the memory and then I set the local name with `BLE.setLocalName()` and the device name with `BLE.setDeviceName()` and here comes the problem : 

The device name works, but the local name doesn't. Even if `bleName` hasn't changed. 
It changes to an unknown character (or two, or sometimes a letter). Even if using `BLE.advertise()`

What am I doing wrong? The read values from the memory are all good.

Thank you!

Quentin

Polldo, 2021-06-09:
I slightly modified the previous sketch in order to test your use case:
``` C++
#include "ArduinoBLE.h"

const char *names[] = {"A", "B", "C"};

void setup()
{
  BLE.begin();
  BLE.setDeviceName("A");
  BLE.setLocalName("Device-Name-Test");
  BLE.advertise();
}

void loop()
{
  static auto timeRef = millis();
  static auto count = 0;

  BLE.poll();
  if (millis() - timeRef > 5000) {
    count++;
    timeRef = millis();
    auto idx = count % 3;
    BLE.setDeviceName(names[idx]);
    BLE.setLocalName(names[idx]);
    BLE.advertise();
  }
}
```

On my side this works correctly.

MCXIV, 2021-06-10:
@Polldo Indeed, this one works. So it's on my side haha.
Sorry for updating this then, and thank you!

Quentin

Polldo, 2021-06-10:
@MCXIV you're welcome !

closed 2021-06-10 by Polldo