arduino-libraries/ArduinoBLE #106: Memory leak in HCICordioTransportClass

Wenn0101, 2020-08-14:
I am working on porting this library to a new target, and I have found what I believe to be a memory leak in the HCICordioTransportClass::write() function.

```
size_t HCICordioTransportClass::write(const uint8_t* data, size_t length)
{
  if (!_begun) {
    return 0;
  }

  uint8_t packetLength = length - 1;
  uint8_t packetType   = data[0];

#if CORDIO_ZERO_COPY_HCI
  uint8_t* packet = (uint8_t*)WsfMsgAlloc(max(packetLength, MIN_WSF_ALLOC));

  memcpy(packet, &data[1], packetLength);

  return CordioHCIHook::getTransportDriver().write(packetType, packetLength, packet);
#else
  return CordioHCIHook::getTransportDriver().write(packetType, packetLength, (uint8_t*)&data[1]);
#endif
}
```

If CORDIO_ZERO_COPY_HCI is defined, memory allocated and never freed.  Locally I have changed this section of code to be:

```
size_t HCICordioTransportClass::write(const uint8_t* data, size_t length)
{
  if (!_begun) {
    return 0;
  }

  uint8_t packetLength = length - 1;
  uint8_t packetType   = data[0];

#if CORDIO_ZERO_COPY_HCI
  uint8_t* packet = (uint8_t*)WsfMsgAlloc(max(packetLength, MIN_WSF_ALLOC));
  if(packet)
  {
    memcpy(packet, &data[1], packetLength);

    uint16_t writtenBytes = CordioHCIHook::getTransportDriver().write(packetType, packetLength, packet);
    WsfMsgFree(packet);
    return writtenBytes;
  }
  else
  {
    return 0;
  }
#else
  return CordioHCIHook::getTransportDriver().write(packetType, packetLength, (uint8_t*)&data[1]);
#endif
}
```

This memory leak was causing hard faults on my target after the buffer pool was full and the alloc returned a null pointer. Is there a workaround for this? I would gladly create a PR should it be decided this is a real issue that could affect other targets.

Polldo, 2020-08-20:
Hi @Wenn0101 ,
when using `CORDIO_ZERO_COPY_HCI`, the ownership of the buffer containing the message is passed to the controller. 
So you should not worry about freeing the allocated memory because it will be handled by the controller.
https://github.com/ARMmbed/mbed-os/blob/6a244d7adffc0e93872cfc880e539ee11bbc6002/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/NRFCordioHCITransportDriver.cpp#L49-L50

Wenn0101, 2020-08-20:
Hi @Polldo Thanks for the response.

Taking a closer look, I am beginning to understand more the purpose of this CORDIO_ZERO_COPY_HCI define, and that it has a matching #define in the mbed core.

I guess my problems come back to defining this when the target I am using does not utilize this in the mbed core.  Right now it seems to me that this CORDIO_ZERO_COPY_HCI is used to gaurd enough in the arduino BLE library that it

Wenn0101, 2020-08-20:
misclick closing, I would appreciate feedback on my comment.

Polldo, 2020-08-28:
@Wenn0101 you are right, at the moment `HCICordioTransport` is intended to work only with `HCI_ZERO_COPY` defined.
If you need to work without `HCI_ZERO_COPY` check this out https://github.com/ARMmbed/mbed-os-cordio-hci-passthrough/pull/2
Basically the packets are inserted in a static buffer rather than being encapsulated in a dynamically allocated wsf_msg.

Polldo, 2020-09-04:
@Wenn0101 I would like to be more precise. Our `HCICordioTransport` actually supports two kind of operation: 
- The first consists in using the cordio controller for managing the ble radio. In this case ArduinoBLE, as a host library, communicates with the controller through a fake uart channel. Here the HCI_ZERO_COPY plus a thread for dispatching ble events are needed. This method is used for the nano33ble board.
- The second consists in bypassing also the cordio controller. In this case, cordio is used just for managing the uart channel with the ble module and other physical connections. So, here HCI_ZERO_COPY and the thread are not needed, the cordio write and receive functions are used directly to forward packets to/from the uart channel. This method is used for the portenta h7 board.

Wenn0101, 2020-09-04:
@Polldo Thanks for the clarification.

I think my problem really stems from the fact that I am porting a target that:
-uses the mbed CORDIO stack to manage an onboard radio (the driver uses wsf events)
-does not utilize the CORDIO_LL in mbed (and thus does not use HCI_ZERO_COPY) (the on-chip ble controller accept a HCI-like data interface)  

This left me wanting to utilize the bleloop as if I was part of your first group, but the target was not configured to free the HCI messages as if i was in the latter group.
I decided to work around this by simply enabling HCI_ZERO_COPY in mbed, and having my driver free the buffer after it passes the data to the hardware.

This is a suitable workaround, and we can close this issue if you would like.

I think in my mind it is still lingering that the macro HCI_ZERO_COPY is being used to determine whether the host and controller are combined on a single chip solution, but there can exist cases where an single chip solution does not want to utilize HCI_ZERO_COPY.

If you are curious the mbed target I am porting can be found [here](https://github.com/sparkfun/mbed-os-ambiq-apollo3/tree/ambiq-apollo3-arduino).

Polldo, 2020-09-07:
@Wenn0101 your workaround sounds good to me. 

>  there can exist cases where an single chip solution does not want to utilize HCI_ZERO_COPY

I agree with you, supporting this use case could be a future enhancement.
However nice porting!
I'm going to close this issue, but feel free to open another one if you happen to have some question or request. 
Thanks!

closed 2020-09-07 by Polldo