arduino-libraries/ArduinoBLE #113: Delays in code execution

Dimedrols, 2020-09-14:
Board: Arduino Nano 33 BLE
Arduino IDE version 1.8.13
Arduino nRF528x Boards version 1.1.6
ArduinoBLE version 1.1.3

After updating the ArduinoBLE library (version 1.1.3) and core for the Arduino Nano 33 BLE (version 1.1.6), my code began to slow down a lot.
When installing older versions of ArduinoBLE library (version 1.1.2) and core (version 1.1.3), everything works fine.
I made a simple example and found out that the code slows down after turning on Bluetooth. Sometimes there is a very strong delay when reading an analog pin or data from a sensor. This delay is clearly visible on the Serial Plotter.

`
#include <ArduinoBLE.h>

// BLE Battery Service
BLEService batteryService("180F");
// BLE Battery Level Characteristic
BLEUnsignedCharCharacteristic batteryLevelChar("2A19",  BLERead | BLENotify);

long beforeMicros = 0;
long afterMicros = 0;

int Value1 = 0;
int Value2 = 0;
int Value3 = 0;
int Value4 = 0;
int Value5 = 0;

void setup() 
{
  // put your setup code here, to run once:

  Serial.begin(9600);

  
  if (!BLE.begin()) 
  {
    while (1);
  }
  
  BLE.setLocalName("BatteryMonitor");
  BLE.setAdvertisedService(batteryService);
  batteryService.addCharacteristic(batteryLevelChar);
  BLE.addService(batteryService);

  BLE.advertise();  
}

void loop() 
{
  // put your main code here, to run repeatedly:

  beforeMicros = micros();
  
  Value1 = analogRead(A0);
  Value2 = analogRead(A1);
  Value3 = analogRead(A2);
  Value4 = analogRead(A3);
  Value5 = analogRead(A4);

  afterMicros = micros();    
  Serial.println(afterMicros - beforeMicros);
}
`

Result:
152
183
214
214
214
214
213
213
183
183
152
183
183
**1251 V**
183
214
214
213
183
183
183
183
213
244
213
214
214
213
213
184
153
184
213
214
213
214
214
214
244
153
214
183
214
213
214
244
214
213
244
183
213
153
183
184
153
183
213
213
214
183
184
214
183
213
244
183
214
213
244
183
152
183
152
214
244
214
183
183
183
213
184
183
244
183
183
183
183
**1251 V**
183
214
214
214
214
183
213
183
183
213
214
213
214
213
183
214
183
214
183
213
213
213
**1281 V**
214
152
183
183

rmlearney-digicatapult, 2020-09-15:
Is this all your code? Where are you writing out the analog values over serial? Why only 9600 baud?

Dimedrols, 2020-09-16:
Yes. This is a simple example to demonstrate the delay.
In this example, I am not using analog values.
For 115200 baud, the result is the same. But for the Arduino Nano 33 BLE it doesn't seem to matter.

Polldo, 2020-09-16:
Hi @Dimedrols , thanks for reporting. 
This behavior does not seem strange to me. The BLE controller runs on a separate thread. Thus, when the BLE is enabled, its thread can take control of the CPU, causing the thread executing the sketch to sleep for a longer time. 
In ArduinoBLE version 1.1.2 this behavior was not present because of an issue: the timer regulating the sleep time of the BLE thread was not started https://github.com/arduino-libraries/ArduinoBLE/commit/a74270d072e056a7c68a5445ba00e1e4b78ec151 , so the BLE controller was always going to sleep and some events were not caught, like the disconnection event.

Dimedrols, 2020-09-16:
This is very bad.
Now you can not receive data from an accelerometer with a sampling rate of 476 Hz or more. I wanted to connect another accelerometer to the board that works with a sampling rate of more than 476 Hz.
I want to draw your attention to the fact that the accelerometer is the only built-in sensor that the Arduino Nano 33 BLE has. And probably many people use it. And the new version of the library has halved the maximum sampling rate.
Now it is clear that this influenced the reading of the analog pin and the reading of data from the memory of the sensors, but it may have affected other processes (for example, the rate of change of the digital pin value).

> This behavior does not seem strange to me. The BLE controller runs on a separate thread. Thus, when the BLE is enabled, its thread can take control of the CPU, causing the thread executing the sketch to sleep for a longer time.

It seems very strange to me.
Why is one thread blocking another. It works in a separate thread so as not to stop the main thread.

> In ArduinoBLE version 1.1.2 this behavior was not present because of an issue: the timer regulating the sleep time of the BLE thread was not started a74270d , so the BLE controller was always going to sleep and some events were not caught, like the disconnection event.

This was done by a very wise man. How can this timer be turned off now?

rmlearney-digicatapult, 2020-09-16:
Happy to test out your findings @Dimedrols - could you possibly post your full code?

facchinm, 2020-09-23:
Hi @Dimedrols 

> This is very bad.

The Nordic chip only has one CPU, so you must use some CPU time to perform other tasks that may be hidden in other concurrent threads.
Pretending that "delays" are always met is a false belief about RTOS. They assure you that your task will be executed before a given deadline, but this deadline cannot be smaller than an OS tick (which is 1ms in mbed).
The BLE stack must periodically perform some work even if you just call `begin()` to ensure its functionality. This is why https://github.com/arduino-libraries/ArduinoBLE/commit/a74270d072e056a7c68a5445ba00e1e4b78ec151 was introduced.

closed 2020-09-23 by facchinm

logotype, 2021-12-08:
This timer.start(); issue in ArduinoBLE is really causing **serious** performance issues. In my case, I'm using the Nano 33 BLE Sense with a stepper motor, requiring smooth continuous motor motion. Tried Scheduler to run BLE/Stepper in separate loops, nope. Also tried the native MBED OS threads, nope. Tried a custom library for hardware interrupts, nope.

The only fix was to patch the library and remove the timer.start(), then it works as expected!

wpilgri, 2023-02-05:
So what is the "proper" design pattern to used when reading a sensor and having ArduinoBLE bluetooth in the same program?  I have found no workaround to the issue described above and changing HCICordioTransport.cpp and removing "timer.start () had no measurable impact.  
[image]

Dimedrols, 2023-02-21:
I'm glad I'm not the only one with this problem. Now there is hope that those who spoil the working code with each new version of the library will pay attention to this problem. After ArduinoBLE version 1.1.3 they "improved" something else, so now removing "timer.start();" was not enough to solve the problem. The problem is that the "improvement" of this library is done by people who do not work in the Arduino company, who solve only their own tasks and do not care about the work of the Arduino Nano 33 BLE for other tasks. If it were employees of the Arduino company, it would be possible to write to the management of Arduino to fire them. But since Arduino is a very poor company that has no money to support libraries for their own boards, there is no one to deal with the normal development of the library.

The best solution is to install old versions of ArduinoBLE Library (ver. 1.1.2) and "Arduino Mbed OS Boards [DEPRECATED!!!]" Board (ver. 1.1.3). Fortunately, the developers of the Arduino IDE took into account such options for "improving" the libraries and made it possible to install old versions of the libraries. Version selection next to the install button. I'm using the old Arduino IDE version 1.8.15, perhaps the Arduino IDE has also been "improved" and removed the ability to select old versions of libraries. If someone checks how this works on the newest version of the Arduino IDE, please write.