arduino-libraries/ArduinoBLE #114: BLE connection from Arduino Hardware not working with latest Windows build19041

arpita-agrawal, 2020-09-15:
I uploaded the example sketch LED available with the ArduinoBLE library on Nano 33 BLE Sense.
After that, I tried to connect the device to the windows system using Bluetooth LE explorer. It doesn't connect. Before updating windows to the latest  build19041 version the connection was working and I could send and receive the data using Bluetooth LE explorer.

Please note that other BLE devices are successfully connecting to the system via Bluetooth LE explorer even after the windows update.

The serial monitor log shows that the board gets connected and disconnected to the PC again and again.

Please refer to the attached screenshots for more details.

Thanks
Arpita 
[image]
[image]
[image]
[image]
[image]

Polldo, 2020-09-16:
Hi @arpita-agrawal ,
this issue should be fixed here https://github.com/arduino-libraries/ArduinoBLE/pull/95
You could checkout the latest commit of ArduinoBLE and try again

arpita-agrawal, 2020-09-16:
Hi,

With the fix in #95 the issue is resolved. However, this fix is not available in version 1.1.3 of library.

Thanks
Arpita

Polldo, 2020-09-16:
Thanks for the feedback, It will be available in the next release!

closed 2020-09-16 by Polldo

ablankenship10, 2021-01-07:
I'm not sure that this is working even with the latest from Master branch. Running the BatteryMonitor example and attempting to connect with BLE Explorer results in the same issue above (Unable to pair, however no Catastrophic error). It is able to connect and starts printing the battery level value to the console for about 20 seconds before the Central (windows pc) disconnects and BLE Explorer shows pairing failed.

Windows version:
[image]

epice500, 2021-01-20:
> I'm not sure that this is working even with the latest from Master branch. Running the BatteryMonitor example and attempting to connect with BLE Explorer results in the same issue above (Unable to [...]

I've been having a similar problem. I am working on a BLE device for windows 10 which was working fine last summer, however when I came back to it in December it no longer worked. I traced the issue back to a windows 10 update released sometime around September - October which breaks the functionality of BLE in windows 10... again. I am currently working on a solution as this is a huge setback and Microsoft seems to be blissfully uncaring about the issue.