arduino-libraries/ArduinoBLE #116: Allow more than 3 concurrent peripheral connections in central mode

muhkuhgh, 2020-09-25:
The nRF52840 used in the Nina B306 board used by e. g. the arduino nano sense can natively support 8 connections and up to 20 (tested up to 16) by changing ram partitions and using softdevice 140

What is the bottleneck allowing only 3 connections here? Any simple way to increase that number at least to 4 or 5?

Polldo, 2020-09-28:
Hi @muhkuhgh ,
the maximum number of connections is set in the mbed configuration of the board. See https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/40 to understand what you should change to increase it.

iactiva, 2020-10-07:
Hello.
I'm working on a MKR 1010 project and I'm stuck on the 3 concurrent peripherals limitation. I've already checked the mbed config solution for nano33ble however mkr1010 is using NINA-10W (ESP32) opposite to NINA-B3 (nRF52840).
For this project I'll need as much concurrent peripherals as possible... any workaround, ideas or suggestions would be welcome.
Thank you

iactiva, 2020-10-22:
I couln't find any solution for increasing the number of 3 concurrent connections working as central on MKR1010 so I gave up this board by now and focus on NANO33BLE. 

I'm following @Polldo work with nRF core for multiconnection [here](https://github.com/Polldo/ArduinoCore-nRF528x-mbedos/tree/ble-multiconnection) as described [here](https://github.com/arduino-libraries/ArduinoBLE/issues/108) and [here](https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/40).

giulcioffi, 2020-10-22:
Hi @iactiva, regarding the multi-connection issue, you could try to use @Polldo 's ArduinoBLE branch [multi-connection](https://github.com/Polldo/ArduinoBLE/tree/multi-connection).
Then, about your issue with MKR WiFi 1010 as central device, you could try this path [here](https://github.com/giulcioffi/nina-fw/tree/issue116BLE). I modified the defines related to the number of maximum BLE connections in the NINA fw. You will then need to flash on the MKR the new fw.
First of all, follow the guide that you find in the [README](https://github.com/giulcioffi/nina-fw/blob/master/README.md). 
Then, to apply the new changes, you need to do the following:
- upload on MKR WiFi 1010 the `SerialNINAPassthrough.ino`
- flash the compiled firmware by running the command proposed in the shell after the fw compilation (remember to select the correct port and to set the baud rate to 921600). The command to flash the fw would be like this (obviously adjusted with your local paths):
`python /home/.../esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyACM1 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /home/.../nina-fw/build/bootloader/bootloader.bin 0xf000 /home/.../nina-fw/build/phy_init_data.bin 0x30000 /home/.../nina-fw/build/nina-fw.bin 0x8000 /home/.../nina-fw/build/partitions.bin`

Once this is done, you can test your multiple connection.
Please let me know if this helps you :)

iactiva, 2020-10-23:
Thank you so much for your solution @giulcioffi !!
Just updated the WiFiNINA firmware and it is now able to connect to 4 peripherals (not yet tested with more as I only have 4 devices by hand right now).
I'll continue testing and post feedback here.

iactiva, 2020-10-28:
After some hours testing hard on the MKR WiFi 1010 board with the updated WiFiNINA FW, it works just perfect for conneting to 4 peripherals however I'm getting about a 25% failure connections to the 5th peripheral and over a 60% failure connections to the 6th peripheral. Have not yet tested with 7 devices yet.
My tests: Scan all available devices and connect to certain mac addresses then read some GATT characteristics and start over scanning.
@giulcioffi Have you been able to get stable connection with 7 devices repeatedly?

closed 2020-10-29 by facchinm