arduino-libraries/ArduinoBLE #127: Missing house keeping logic for ATT class

dreamstyler, 2020-10-17:
The ATT class is missing an end() function to clean up ATT variables. This end function then needs to be called inside BLELocalDevice::end().

Proposed function:
void ATTClass::end()
{  
  for (int i = 0; i < ATT_MAX_PEERS; i++) {
    if (_peers[i].connectionHandle == 0xffff) {
      continue;
    }
    _peers[i].connectionHandle = 0xffff;
    _peers[i].role = 0x00;
    _peers[i].addressType = 0x00;
    memset(_peers[i].address, 0x00, sizeof(_peers[i].address));
    _peers[i].mtu = 23;

    if (_peers[i].device) {
      delete _peers[i].device;
      _peers[i].device = NULL;
    }
  }
}

I set up my BLE as a peripheral and use it to stream data to an android device. The mobile requests data using an index (writes to a characteristic) and then my IoT33 nano has a data characteristic which it updates. The mobile app listens for change notifications on the data characteristic, and once it read it requests the next index. I am using a MTU of 242.
This works ok but the Arduino seems to crash at some random points. From 10000 data requests it can crash at any point.

To address this I restart the BLE module by calling end() and then being().
I noticed that sometimes the MTU from the peers in ATT were not being cleared.
After further investigation, I noticed there is some house keeping missing in ATT end() to clean all the _peers.

Not sure if this is an issue, but disconnect() should probably clean up all _peers even if **HCI.disconnect(_peers[i].connectionHandle)** does not return 0;

Polldo, 2020-10-19:
Hi @dreamstyler .
Thanks for reporting. 
Rather than resetting the BLE stack, I would try to detect the cause of your 'crash at some random point'. 
Could you post the sketch you are using and attach the debug information printed into the serial monitor? (To print the debug information you should insert this statement `BLE.debug(Serial);` after the begin of the serial monitor in your sketch).

However, thanks also for this report https://github.com/arduino-libraries/ArduinoBLE/issues/129 . I will address it in the coming days!

dreamstyler, 2020-10-20:
Thanks for your reply,

I will check later to give some sample code. 

Please find below a log dump
> Log 19B10001-0001-537E-4F6C-D104768A1214 characteristic event written bytes: 4
   is when the android app "request" a new chunk of data using a index value
  this triggers a call-back in Arduino on the characteristic which sets a flag for a data update
> BLE load log data idx: 1764
  on the next processing loop the Arduino will check the update flag and update the data characteristic which will NOTIFY the android app to read the characteristic. (loads 480 bytes)
> I am aware of the MTU and that not the full 480 bytes are sent. Did have test with 200 bytes as well, but some end result.
**NOTE: It would be good to have a way to check what MTU was negotiated with the other BLE device as this informs one on how much data can be transferred. (this is stored on peers but there is no where to check what that is)** 

There are a few cases where this below happens, but id does seem to keep working
HCI EVENT RX <- 04130004
13
5
1
0
0
1
0

At there points is when coms stops working and a reboot is required. Its as if the Ardrino BLE module freezes and does not accept new requests from other BLE device. Did try query the BLE mac address of the Arduino at this point and it return 0's.
Reboot is where I trigger a reboot.
> READ ERROR
> REBOOTING SYSTEM:

`HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200B0007000400120300E4060000
Log 19B10001-0001-537E-4F6C-D104768A1214 characteristic event written bytes: 4
HCI ACLDATA TX -> 02000005000100040013
BLE load log data idx: 1764 HCI ACLDATA TX -> 020000F600F20004001B0500E40600001E4DE81F2705000101010101FF000000214DE81F270696019701970195010000B94DE81F27078622AC22CE22682200009189E81F2705000100010101FF0000004D89E81F2706960198019801950100004889E81F270782222A22CE222A2200004BC5E81F2705000100010101FF00000010C5E81F270697019601980196010000EAC5E81F270781228322CE222A2200000E01E91F2705000100010101FF0000001301E91F270697019701980196010000AA01E91F27079822EA22EA222A2200004B3DE91F270500010101010100010000073DE91F270697019701980196010000BD3DE91F27078F229822EA
 | bytes loaded: 480
HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200B0007000400120300F2060000
Log 19B10001-0001-537E-4F6C-D104768A1214 characteristic event written bytes: 4
HCI ACLDATA TX -> 02000005000100040013
BLE load log data idx: 1778 HCI ACLDATA TX -> 020000F600F20004001B0500F20600001E3DE91F27078F229822EA222A2200007B79E91F2705000101010101000100003E79E91F270697019601980196010000C779E91F27078D229E22EA222A22000054B5E91F27050001010101010001000075B5E91F27069601960197019601000066B5E91F270793226122EA225A22000096F1E91F2705000100010101000100000FF1E91F27069601950197019501000012F1E91F27079822B322EA225A220000CB2DEA1F270501010101010100010000752DEA1F270696019601970195010000202DEA1F270786225322D52253220000E969EA1F2705010101010101000100004C69EA1F27069601960197
 | bytes loaded: 480
HCI EVENT RX <- 04130004
13
5
1
0
0
1
0
HCI ACLDATA RX <- 0200200B000700040012030000070000
Log 19B10001-0001-537E-4F6C-D104768A1214 characteristic event written bytes: 4
HCI ACLDATA TX -> 02000005000100040013

BLE load log data idx: 1792 HCI ACLDATA TX -> 020000F600F20004001B0500000700001E69EA1F2706960196019701950100001969EA1F270777223822D5221C220000C5A5EA1F27050101010101010001000007A5EA1F27069601960196019501000065A5EA1F27078022DC22DC221C22000089E1EA1F2705010101010101000100003EE1EA1F27069601950196019501000099E1EA1F270780228322DC221C220000571DEB1F270501010101010101010000971DEB1F2706960196019601950100007A1DEB1F270777228322DC221C2200008659EB1F270501010101010101010000AE59EB1F2706960196019701950100007459EB1F270771229822DC221C2200009295EB1F27050101010101
 | bytes loaded: 480
HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200B00070004001203000E070000
Log 19B10001-0001-537E-4F6C-D104768A1214 characteristic event written bytes: 4
HCI ACLDATA TX -> 02000005000100040013
BLE load log data idx: 1806 HCI ACLDATA TX -> 020000F600F20004001B05000E0700001E95EB1F270501010101010101010000E595EB1F2706960194019701940100003695EB1F270781223F22DC223F22000064D1EB1F270501010101010101010000DCD1EB1F27069501930197019301000012D1EB1F27077D229822B3223F220000CA0DEC1F2705010101010101010100009B0DEC1F270695019301970193010000550DEC1F27078222BA22BA223F220000E449EC1F270501010101010101010000A249EC1F2706940192019701920100006E49EC1F270786225A22BA223F220000BD85EC1F270501010101010101010000E985EC1F2706930191019601910100000785EC1F27079422EA22EA
 | bytes loaded: 480
HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 0413050100000100
HCI ACLDATA RX <- 0200200B00070004001203001C070000
Log 19B10001-0001-537E-4F6C-D104768A1214 characteristic event written bytes: 4
HCI ACLDATA TX -> 02000005000100040013
BLE load log data idx: 1820 HCI ACLDATA TX -> 020000F600F20004001B05001C0700001E85EC1F27079422EA22EA223F2200007FC1EC1F270501010101010101010000D0C1EC1F27069301910194019101000050C1EC1F27079A229E22EA225A22000047FDEC1F270501010101010101010000C7FDEC1F270692019201930191010000C9FDEC1F27079C22B322EA225A2200001239ED1F270501010101010101010000C439ED1F270692019001930190010000C339ED1F2707A222F722F7225A2200003875ED1F2705010101010101010100009975ED1F27069101920193018F010000F175ED1F2707AA22CE22F7226822000099B1ED1F270501010101010101010000B6B1ED1F27069101910193
 | bytes loaded: 480
[... 463 lines cut ...]
 | bytes loaded: 480
HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 0413050100000100
[... cut ...]
BLE load log data idx: 2534 HCI ACLDATA TX -> 020000F600F20004001B0500E6090000056A090000062D012E012F012D010000946A09000007130012001300120000005B960A0000051C011D011D011C010000EB960A0000063001310132012F010000E5960A00000713001300130013000000E900000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
 | bytes loaded: 80
HCI EVENT RX <- 0413050100000100
HCI EVENT RX <- 0413050100000100`

dreamstyler, 2020-10-20:
Added a sample project, the actual project is much larger.
I use FreeRTOS so I had to "unpluck some functions" to add vTaskSuspendAll, xTaskResumeAll, vTaskDelay
Also commented out some code, like reading data from SD card, but this should give you an idea of what I have done thus far.

[BlueTest.zip](https://github.com/arduino-libraries/ArduinoBLE/files/5408649/BlueTest.zip)