arduino-libraries/ArduinoBLE #128: On Nano 33 BLE Sense - Writing to Characteristic via BLE gets stuck periodically

hurrycane, 2020-10-17:
Hi Everyone,

I'm trying to put together a simple sketch that connects to a given BLE service UUID and sends data to a specific characteristic (also given by UUID). What I've observed is that most often the ArduinoBLE client get stuck periodically or enters a bad state. This only happens on the Nano 33 BLE Sense board but it doesn't happen on the Portenta H7 board.

Here's the sketch (heavily based on the LedControl example sketch)

```
#include <Arduino.h>
#include <ArduinoBLE.h>

String deviceUuid = "795090c7-420d-4048-a24e-18e60180e23c";
char const * characteristicUuid = "7069e4d8-0ab2-11eb-adc1-0242ac120002";

void setup() {
  Serial.begin(115200);
  while(!Serial);

  BLE.debug(Serial);

  // initialize the BLE hardware
  if (!BLE.begin()) {
     while (1);
  }

  Serial.println("BLE Central - LED control");

  // start scanning for peripherals
  BLE.scanForUuid(deviceUuid);
}

void loop() {
  // check if a peripheral has been discovered
  BLEDevice peripheral = BLE.available();

  if (peripheral) {
    // discovered a peripheral, print out address, local name, and advertised service
    Serial.print("Found ");
    Serial.print(peripheral.address());
    Serial.print(" '");
    Serial.print(peripheral.localName());
    Serial.print("' ");
    Serial.print(peripheral.advertisedServiceUuid());
    Serial.println();

    if (peripheral.localName() != "BV5900") {
      return;
    }

    // stop scanning
    BLE.stopScan();

    controlLed(peripheral);

    // peripheral disconnected, start scanning again
    BLE.scanForUuid(deviceUuid);
  }
}

void controlLed(BLEDevice peripheral) {
  // connect to the peripheral
  Serial.println("Connecting ...");

  if (peripheral.connect()) {
    Serial.println("Connected");
  } else {
    Serial.println("Failed to connect!");
    return;
  }

  // discover peripheral attributes
  Serial.println("Discovering attributes ...");
  if (peripheral.discoverAttributes()) {
    Serial.println("Attributes discovered");
  } else {
    Serial.println("Attribute discovery failed!");
    peripheral.disconnect();
    return;
  }

  // retrieve the LED characteristic
  BLECharacteristic ledCharacteristic = peripheral.characteristic(characteristicUuid);

  if (!ledCharacteristic) {
    Serial.println("Peripheral does not have LED characteristic!");
    peripheral.disconnect();
    return;
  } else if (!ledCharacteristic.canWrite()) {
    Serial.println("Peripheral does not have a writable LED characteristic!");
    peripheral.disconnect();
    return;
  }

  if (peripheral.connected()) {
    // while the peripheral is connected
    ledCharacteristic.writeValue((byte)0x01);
  }

  Serial.println("Peripheral disconnected");
}```

The way this gets stuck is the following:
```
16:09:34.707 -> BLE Central - LED control
16:09:34.845 -> Found 77:d3:6f:9f:21:5d 'BV5900' 795090c7-420d-4048-a24e-18e60180e23c
16:09:34.845 -> Connecting ...
16:09:35.371 -> Connected
16:09:35.371 -> Discovering attributes ...
16:09:35.475 -> Attribute discovery failed!
16:09:35.510 -> Found 55:2c:ba:72:dd:93 'BV5900' 795090c7-420d-4048-a24e-18e60180e23c
16:09:35.510 -> Connecting ...
16:09:35.792 -> Connected
16:09:35.792 -> Discovering attributes ...
16:09:35.894 -> Attribute discovery failed!
16:09:36.067 -> Found 60:a0:ea:c1:18:56 'BV5900' 795090c7-420d-4048-a24e-18e60180e23c
16:09:36.067 -> Connecting ...
16:09:36.338 -> Connected
16:09:36.338 -> Discovering attributes ...
16:09:36.439 -> Attribute discovery failed!
16:09:36.475 -> Found 77:d3:6f:9f:21:5d 'BV5900' 795090c7-420d-4048-a24e-18e60180e23c
16:09:36.510 -> Connecting ...
16:09:37.026 -> Connected
16:09:37.026 -> Discovering attributes ...
16:09:37.127 -> Attribute discovery failed!
16:09:37.195 -> Found 67:a4:ec:39:e4:40 'BV5900' 795090c7-420d-4048-a24e-18e60180e23c
16:09:37.195 -> Connecting ...
16:09:37.715 -> Connected
16:09:37.715 -> Discovering attributes ...```

I've tried multiple things including different Arduino Nano 33 BLE Sense boards. You can find attached the output I get from the H7 board and the output I get while enabling the debug mode for the BLE library.

Help is much appreciated!
Thanks!
B.
[nano33_ble_stuck.txt](https://github.com/arduino-libraries/ArduinoBLE/files/5396861/nano33_ble_stuck.txt)
[h7_normal_output.txt](https://github.com/arduino-libraries/ArduinoBLE/files/5396862/h7_normal_output.txt)
[nano33_ble_sense_debug_call_log.txt](https://github.com/arduino-libraries/ArduinoBLE/files/5396863/nano33_ble_sense_debug_call_log.txt)
[nano33_ble_stuck_debug.txt](https://github.com/arduino-libraries/ArduinoBLE/files/5396864/nano33_ble_stuck_debug.txt)

Polldo, 2020-10-19:
Hi @hurrycane , thanks a lot for reporting!
This issue is similar to one I've solved in the past by adding this patch to mbed-os https://github.com/Polldo/ArduinoCore-nRF528x-mbedos/commit/61b95ab8cb03f2bf8d57ba62ac25f807e209ad7d
Would you mind trying again the nano33ble with this version of the core? https://github.com/Polldo/ArduinoCore-nRF528x-mbedos/tree/ble-multiconnection
Thanks!

RMLEARNEY, 2021-03-10:
@Polldo is that patch going to be merged in future?

Polldo, 2021-03-11:
@RMLEARNEY yes, it's going to be included in the new release of the mbed core