arduino-libraries/ArduinoBLE #130: [mbed boards] Packet drops cause failures

Polldo, 2020-10-26:
**Background**
Mbed boards (nano33ble, portentaH7) running ArduinoBLE have two relevant concurrent threads in execution: one executes the main arduino sketch, the other one handles the transport of HCI packets and stores the received ones in an internal buffer. 
The thread executing the sketch should regularly call the `BLE.poll()` function to retrieve and analyze the packets stored by the other thread.

**Problem:**
If too much time passes between calls to the `BLE.poll()` function, then the buffer of received packets may become full. If it becomes full then the new packets received will be dropped. (see [code](https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/HCICordioTransport.cpp#L288-L291))
This behaviour may result in a buffer overflow and consequently in a system crash.

**Possible solutions**

- Discourage the use of blocking functions (`delay(millis)` should be avoided. An alternative `BLE.delay()` that calls `BLE.poll()` could be implemented)
- Handle packets drop to prevent buffer overflows
- Increase the reception buffer to increase reliability in case some blocking function is essential.
- Stop the controller until some space is freed from the buffer (see [#96](https://github.com/arduino-libraries/ArduinoBLE/pull/96))  

**Related issues**
https://github.com/arduino-libraries/ArduinoBLE/issues/126
https://github.com/arduino-libraries/ArduinoBLE/issues/73
https://github.com/arduino-libraries/ArduinoBLE/issues/102

andbe-gh, 2020-10-27:
Hi @Polldo,
the changes in HCICordioTransport.cpp (#96) seem to fix my problem (#126) with hanging arduino when many BLE devices are nearby. I've succesfully tested it with 32 devices.

andbe-gh, 2021-03-17:
Is the problem fixed in the new release? I can't find anything in the changelog from 1.1.3 to 1.2.