arduino-libraries/ArduinoBLE #142: Arduino BLE not working with Artemis Arduino_Apollo3 Core v2.0.2

pratapppv, 2020-11-17:
The ArduinoBLE library isn't working with Sparkun Artemis module when using Arduino_Apollo3 Core v2.0.2. The library seems to be stuck in an infinite loop when the BLE.begin() function is called. The board I am using is the Sparkfun Artemis Thing Plus.

facchinm, 2020-11-17:
@pratapppv thanks for reporting. Are you using ArduinoBLE from git or the released version?

pratapppv, 2020-11-17:
Hey, I am using it from git.

facchinm, 2020-11-17:
So that's strange... [This](https://github.com/arduino-libraries/ArduinoBLE/commit/de44016689091c99425f671082ba65495f5e962a#diff-1a79f029352e98b813aaf6a97ee51cf595503c6e75ae51cb4a04a5f705db9853) is the only commit regarding Cordio that was merged recently (between 1.1.3 and `master`). Could you bisect the tree to get the offending patch? I'd do it but I don't have an Apollo board to test it.
@Wenn0101 could you help?
Thanks!

pratapppv, 2020-11-17:
@facchinm I'll do so and keep you posted(might take me a day or so)

Ragiton, 2020-11-19:
Sorry if this isn't the best way to link the part in the code where I'm having an issue. I'm not very experienced with github.
Anyway, If you comment out the line [here](https://github.com/arduino-libraries/ArduinoBLE/commit/d435309d3da0a3b804e085b905e1a2152f680751#r44325202) the ble starts working again.

Wenn0101, 2020-11-19:
@facchinm @Ragiton thanks for taking a look at this.
We got a bug report on this in our arduino repo, and it may be a problem with our latest release.

I am finishing up some other bugs and I can start contributing to this issue.

facchinm, 2020-11-19:
@Ragiton great find! That line might be ineffective on nano33ble and is probably a leftover of the "common path" with Portenta.
@Polldo would you mind taking a look and eventually removing it?

Polldo, 2020-11-19:
@facchinm I gave a look at the mbed `NRFCordioHCIDriver` implementation. It seems that the `start_reset_sequence()` function is used by the cordio **host** in order to reset the controller ([ref.](https://github.com/ARMmbed/mbed-os/blob/master/connectivity/drivers/ble/FEATURE_BLE/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/NRFCordioHCIDriver.cpp#L327-L333)).
Since we are using only the cordio **controller** and given that we already send a reset command to the controller [(here)](https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/local/BLELocalDevice.cpp#L90), I think we can remove that line.
I'll test it and let you know