arduino-libraries/ArduinoBLE #144: Central is notified when it changes characteristic value

arkadiuszmarszalek, 2020-11-23:
When we use ArduinoBLE (1.1.3) as Peripheral device it notifies subscribed Central of characteristic value change, even if the write came from Central itself.

[notification-fix.patch.zip](https://github.com/arduino-libraries/ArduinoBLE/files/5582245/notification-fix.patch.zip)

cbrunschen, 2023-04-15:
There is a set of fairly popular [BLE to Serial adapters from DSD Tech](https://www.deshide.com/Bluetooth-Module.html) (https://www.deshide.com/) which use a service with UUID `FFE0` and a single characteristic `FFE1` for both reading and writing data.

I tried to implement the same sort of behaviour on an ESP32. With the Espressif BLE libraries, this is straightforward. Trying to implement the same with ArduinoBLE echoes back everything that the connected Central sends to the ESP32, because of this very issue: the Central is notified of the data that it has just written to the ESP32.

This patch would fix that.

marcosinigaglia, 2023-11-21:
Is there any news about it?

loginov-rocks, 2024-08-15:
I also faced the same problem when using the same characteristic for both writing and notifying using this library:

```
BLEService myService("ffe0");
BLECharacteristic myCharacteristic("ffe1", BLERead | BLEWriteWithoutResponse | BLENotify, 20);
```

I assumed, which seems wrong, that `BLEWriteWithoutResponse` is supposed to avoid echoing. Interestingly, the [ESP32 BLE](https://github.com/nkolban/ESP32_BLE_Arduino) library works without echoing the value written.

Can be reproduced with https://github.com/loginov-rocks/BluetoothTerminal/blob/main/src/main.cpp together with the [Web Bluetooth Terminal](https://loginov-rocks.github.io/Web-Bluetooth-Terminal/) app. And [here's](https://github.com/loginov-rocks/BluetoothTerminal/blob/90dc9cb77fedbd3d6ee7ce2c9ed9172bbc290618/src/main.cpp) implementation using the [ESP32 BLE](https://github.com/nkolban/ESP32_BLE_Arduino) library, which works seamlessly.

Screenshots attached, for every "idle" message the code sends a "Test" message that can be seen in the web app, but echoes what was sent from the web app:

[image]

[image]

I appreciate any ideas on how that can be avoided!

loginov-rocks, 2024-09-16:
Hey guys, checking if there is anything new about this one? Any ideas on how to avoid?