arduino-libraries/ArduinoBLE #149: BLE Power consumption after BLE.end()

oRMMo, 2020-12-11:
Hi,

Related to #65 and the fix #92, BLE.end() still draw around ~0.5 mA on NANO 33 BLE Sense.
 
@facchinm mentioned that the CordioHCIHook doesn't shut down the radio (https://github.com/arduino-libraries/ArduinoBLE/issues/65#issuecomment-653004787).
So I tried inserting  NRF_RADIO->TASKS_STOP = 1; or/and NRF_RADIO->TASKS_DISABLE = 1; after BLE.end() with no success (wishful thinking I believe :)).

This sketch is around ~ 1.562 mA

```

#include <ArduinoBLE.h>

BLEService serv("19B10010-E8F2-537E-4F6C-D104768A1214");
BLEByteCharacteristic charac("747ce3a8-3588-11eb-adc1-0242ac120002", BLERead | BLENotify);

void setup() {
  
  digitalWrite(LED_PWR, LOW);
  
  Serial.begin(9600);
  while(!Serial);
  
  if (!BLE.begin()){
    Serial.print("NO BLE");
    while(1);                                                                        
  } else {
    Serial.println("BLE Started");  
  }
  
  BLE.setLocalName("TEST_BLE_Power");
  BLE.setAdvertisedService(serv);
  serv.addCharacteristic(charac);
  BLE.addService(serv);
  BLE.setAdvertisingInterval(1600);
  BLE.advertise();
  BLE.end();
}

void loop() {
  delay(1000);
}
``` 
The same sketch without the BLE.begin() (without starting then stopping back BLE) is around ~1.067 mA
 ```
#include <ArduinoBLE.h>

BLEService serv("19B10010-E8F2-537E-4F6C-D104768A1214");
BLEByteCharacteristic charac("747ce3a8-3588-11eb-adc1-0242ac120002", BLERead | BLENotify);

void setup() {
  
  digitalWrite(LED_PWR, LOW);
  
  Serial.begin(9600);
  while(!Serial);
  
  /*if (!BLE.begin()){
    Serial.print("NO BLE");
    while(1);                                                                        
  } else {
    Serial.println("BLE Started");  
  }*/
  
  BLE.setLocalName("TEST_BLE_Power");
  BLE.setAdvertisedService(serv);
  serv.addCharacteristic(charac);
  BLE.addService(serv);
  BLE.setAdvertisingInterval(1600);
  BLE.advertise();
  BLE.end();
}

void loop() {
  delay(1000);
}
```
If I run the same sketches without Serial.begin() & no USB cable connected the consumption goes down of course but the BLE.end() "cost" stays the same.

I am trying to get as much low power consumption as possible for my project and I am actually down to around  ~205uA without radio. With radio on only for TX when needed, everything would be set for months on battery. Still need to find a solution for the BLE Off power draw.
I tried digging in the Arduino BLE library but it is way beyond my knowledge & skill.

Any ideas how to go about this? Any help is welcome.

facchinm, 2020-12-14:
Hi @oRMMo ,
we are working closely with arm to bring the low power performance on nRF52 on par with softdevice's. Next version of the nordic core will contain some important fixes. I'll keep you posted as soon as we have some pre-release core if you want to take a look :slightly_smiling_face:

oRMMo, 2020-12-14:
Hi @facchinm ,
Thanks for the heads up! I will more than happy to take a look :thumbsup:

tangtangdn, 2021-02-02:
> Hi @oRMMo ,
> we are working closely with arm to bring the low power performance on nRF52 on par with softdevice's. Next version of the nordic core will contain some important fixes. I'll keep you posted as soon as we have some pre-release core if you want to take a look 🙂

Hi Facchinm, thank you very much for your information. I have the same problem. Using both BLE.begin() and BLE.end() leads to the power consumption increase of about 300uA. However, it should be very close to without BLE.begin after using BLE.end() (I turn on the BLE, then turn off it, I think it should not cost more power after BLE turned off). Could you kindly tell me what caused the power consumption increase after BLE.end? i.e. what is enabled after BLE.begin but not disabled after BLE.end(). Thank you very much.

CaptainFalcon92, 2023-08-22:
Good day. I'm joining the nRF52840 party with the Xiao Seeed BLE today, and i wonder if this consumption issue is still a thing to this day ?

Best,