arduino-libraries/ArduinoBLE #160: Advertising Services and Manufacturer Data (scanCallback)

NgoWun, 2021-02-22:
As discussed in this Arduino Forum thread, [https://forum.arduino.cc/?topic=728591#msg4902344](url), Manufacturer Data is not displayed if services are advertised, possibly because the byte limit is exceeded when both are used.

Based on other information gleaned from the web, perhaps a scanCallback-type event could be used instead.  This would allow another 31 bytes of information to be transferred in the advertising window before connecting with a central device.  One possible suggestion is to modify the current callback function:

  `setEventHandler(BLEScanCallback, _doSendAdditionalAdvertisingPacket)`

These two articles may provide additional information:
[https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2017/02/10/bluetooth_advertisin-hGsf](url)
[https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2017/11/14/bluetooth_advertisin-zCHh](url)

facchinm, 2021-02-23:
@Polldo can you take a look?

Polldo, 2021-02-24:
Hi @FluffyMarshMallow26 ,
>Manufacturer Data is not displayed if services are advertised

This has been fixed by this [PR](https://github.com/arduino-libraries/ArduinoBLE/pull/117). You need to download the library from github because it has not been released yet.

>Based on other information gleaned from the web, perhaps a scanCallback-type event could be used instead. This would allow another 31 bytes of information to be transferred in the advertising window before connecting with a central device. 

Scan response is already enabled and used. Also thanks to the same [PR](https://github.com/arduino-libraries/ArduinoBLE/pull/117), you can now decide whether to put your advertising data (i.e. local name, service uuid, manufacturer data, etc..) in the main advertising packet or in the scan response packet. 
Example [here](https://github.com/arduino-libraries/ArduinoBLE/pull/134/files). 
Moreover, see https://github.com/arduino-libraries/ArduinoBLE/pull/133 , functions used for adding advertising data (i,e. `setLocalName()` etc.) now return a bool value. This value indicates whether the passed parameter has been correctly added to the advertising packet or not. Thus, if you try to insert a parameter that would make the advertising packet exceed the maximum length, that is 31 bytes, you will receive `false` as a return value.

FluffyMarshMallow26, 2021-02-24:
Based on Polldo's comments, this features has already been implemented and will be eventually be released.

Thank you for responding to this request - closing this request 24 Feb 2021.

closed 2021-02-24 by FluffyMarshMallow26