arduino-libraries/ArduinoBLE #168: Error after writing to GATT characteristic (value>400) three-four times

aburehan2020, 2021-03-24:
Hi!

I am using ARDUINO NANO 33 BLE SENSE as peripheral and Web Bluetooth API to write a characteristic value on it.
The value size is 512 bytes.
I am using this sketch: https://github.com/arduino-libraries/ArduinoBLE/tree/master/examples/Peripheral/LED with the following modification:
```c++
//BLEByteCharacteristic switchCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite);
BLECharacteristic switchCharacteristic("19B10001-E8F2-537E-4F6C-D104768A1214", BLERead | BLEWrite, 512);
```
I can successfully write to the characteristic first few times but then it gets stuck.
The issue also occurs if I write to four different characteristics (512 each) one after another.
Occurs even if I disconnect between writes.
From my observation, this issue never occurs for value size<200 bytes. It occurs sporadically for size=256.

Web Bluetooth script to reproduce the issue:
```html
Connect then Write<br>
<button id="connect">Connect</button>
<button id="write1">Write</button>

<script>
var bleService = '19b10000-e8f2-537e-4f6c-d104768a1214'
var bleCharacteristic1 = '19b10001-e8f2-537e-4f6c-d104768a1214'
var arduinoDevice;
var gattService;
var gattCharacteristic1;
const testbyte1= new Uint8Array([0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x41,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x71,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50,
							0x51,0x42, 0x43, 0x44 , 0x45, 0x46,0x47, 0x48, 0x49, 0x4a, 0x4b, 0x4c, 0x4d, 0x4e, 0x4f, 0x50]);
console.log(testbyte1);

function connect(){
	navigator.bluetooth.requestDevice({ filters: [{services: [bleService]}] })
	.then(device => {arduinoDevice= device;
		return arduinoDevice.gatt.connect();})
	.then(server => server.getPrimaryService(bleService))
	.then(service => {gattService= service;
		return gattService.getCharacteristic(bleCharacteristic1)})
	.then(characteristic1 => {gattCharacteristic1= characteristic1;
		console.log("Connect done");})
	.catch(error => { console.error(error);
					Promise.resolve();});
}

function write(){
	if(arduinoDevice.gatt.connected){
		gattCharacteristic1.writeValue(testbyte1.slice(0,512))
		.then(_ =>{
			console.log('Written')
		})
		.catch(error => { console.error(error)
						Promise.resolve();});
	}
	else{ alert('Device not connected');
	}
}

document.getElementById('connect').addEventListener('click', connect, false);
document.getElementById('write1').addEventListener('click', write, false);
</script>
```

Polldo, 2021-05-24:
Hi @aburehan2020 , 
in the BLE stack there is a parameter called **MTU**. This param is device specific. It indicates the maximum amount of bytes that can be transmitted through BLE. 
In the case of the nano33ble, that is mounting an nrf52, the max MTU is 247. Thus, you can only write up to 247 - 3 bytes (due to the ATT overhead)

aburehan2020, 2021-05-24:
Hi @Polldo ,
If that is the case then shouldn't it be that you can never write more than 244 bytes? I can write that about 2-3 times before it fails.
I do not know a lot about microcontrollers. If that really explains it then you can close this issue.

Polldo, 2021-05-26:
@aburehan2020 yes, you cannot receive more than 244 bytes in a single transfer. I'm quite sure you are only receiving the first 244 bytes of the buffer

wpilgri, 2023-01-26:
I am SOOO happy I found this as I have been trying to use my Arduino 33 BLE to gather some sensor data at a high rate and then transfer via bluetooth.  I am trying to capture 100 readings that consume a 24 byte struct per reading and transfer it all via bluetooth to my laptop.  Works great for 10 points (240 bytes) but caps out at 244 bytes as described above.  Despite declarations of burstmodestructDataChar() to the contrary below, it looks like it is ignored.

```
#include <Arduino.h>
#include <ArduinoBLE.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BNO055.h>
#include <utility/quaternion.h>

//----------------------------------------------------------------------------------------------------------------------
// BLE UUIDs
//----------------------------------------------------------------------------------------------------------------------
#define BLE_UUID_ENVIRONMENT                      "181C" //Service

#define BLE_UUID_RUNMODE                          "2A56" //Characteristic for string writing and reading.

#define BLE_UUID_CONTMODE_STRUCTDATA              "2A57" //Characteristic for calibration string writing.
                                                          // As a struct.  Most efficient method vs. a string.

#define BLE_UUID_BURSTMODESTRUCTDATA              "2A58"  //Characteristic for data as as struct in burst mode
                                                          // which will return multiple structs in an array

#define BLE_UUID_NUMPTS                           "2A21" // Number of points to record when in burst mode.  TBD

#define BLE_UUID_BURST_OPMODE                    "2A4E"  // Mode of operation.
                                                          // To be defined:  False=Continuous, True=burst mode.

//----------------------------------------------------------------------------------------------------------------------
// BLE
//----------------------------------------------------------------------------------------------------------------------
#define BLE_DEVICE_NAME                           "Arduino Nano 33 BLE"
#define BLE_LOCAL_NAME                            "SensorMonitor"  //The name that Python uses to find device

//----------------------------------------------------------------------------------------------------------------------
// CONSTANTS
//----------------------------------------------------------------------------------------------------------------------
#define SAMPLE_RATE 1000    // How often do we read and update BLE data.  Delay in ms.

#define NUMPTS_BURSTMODE 100  //Number of points to record when in burst mode

//----------------------------------------------------------------------------------------------------------------------
// STRUCTURE DEFINITION
// Note:  The unpacking in Python needs to match the byte configuration of the struct here.
//----------------------------------------------------------------------------------------------------------------------
    // uint_8 = 1 byte
    // int = 4 bytes
    //unsigned long = 4 bytes
    //unisgned short =u_int16_t = 2 bytes

typedef struct  __attribute__ ((packed)) {
  unsigned long timeread;
  int qw; //float Quaternion values will be scaled to int by multiplying by constant
  int qx;
  int qy;
  int qz;
  uint8_t cal_system;
  uint8_t cal_gyro;
  uint8_t cal_accel;
  uint8_t cal_mag;
}sensordata ;

sensordata d;

//----------------------------------------------------------------------------------------------------------------------
//  #########################   FUNCTION DEFINITIONS.  NEED FOR PLATFORMIO  (not Arduino)      #########################
//
//      https://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

void writeHandler(BLEDevice central, BLECharacteristic characteristic);
[... 327 lines cut ...]
  for (int i=0;i<valuelen ;i++) {
    setval=(u_int8_t)characteristic.value()[i];
    Serial.print(setval);
  }
  Serial.println("");

  if (setval==false)
    {
      //If false -> in stop Mode
      Serial.println("runmode set to FALSE");
    } else
    {
      Serial.println("runmode set to TRUE");
    }

} //End Writehandler
```

So the million dollar question is how are you supposed to transfer a large amount of data over bluetooth?  I need a continuous sampling of data points so I can do a proper FFT, thus the reason I was trying to grab a large number of points.

Any advice on how a large amount (10,000 bytes) of data is most effectively sent over bluetooth??

Where can I find the MTU setting so I can go inspect the code?  I assume there's no opportunity to override this value?

Thank you Very Much...learning something new every day!